openrazer/openrazer #703: Device Not Recognised

tycruickshank, 2018-10-28:
Hello, I'm getting a 'device not recognized' error when running polychromatic. I'm running Fedora 28 and I'm fairly new to Linux. 
[image]

I have a Death Adder Elite and a Black Widow Ultimate.

I have installed openrazer and polychromatic by adding the repository `dnf config-manager --add-repo http://download.opensuse.org/repositories/hardware:razer/Fedora_28/hardware:razer.repo` and by installing openrazer `dnf install openrazer-meta kernel-devel`. I ran them both as root using sudo.

I've looked through a lot of open and closed issues relating to this on github, but I haven't found a solution that works.

When I run `sudo dkms install openrazer-driver/2.4.0`
I get the output
`Module openrazer-driver/2.4.0 already installed on kernel 4.18.16-200.fc28.x86_64/x86_64`

When I run `sudo modprobe razerkbd`
I get
`modprobe: ERROR: could not insert 'razerkbd': Operation not permitted`

Running `openrazer-daemon -Fv`
outputs `Unable to lock on the pidfile.`

Can anybody offer any help? I'm not sure what to do next.

z3ntu, 2018-10-28:
Disable secure boot.

tycruickshank, 2018-10-28:
I'm currently dual booting this machine with windows 10. Can disabling secure boot potentially hose the windows install? When I go to disable it in the UEFI menu I get a warning that I may need to reinstall the OS.

lah7, 2018-10-28:
Disabling secure boot won't be a problem, it's only [re-enabling it](https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/disabling-secure-boot) which may require re-installation of Windows as it warns. There are _advanced_ workarounds to re-enable secure boot without data loss if you wanted to do so later.

The alternate is to sign the DKMS modules in Fedora yourself to fix the `Operation not permitted` error, but we have don't have sufficient documentation on the matter.

tycruickshank, 2018-10-29:
Thanks for getting back to me regarding this. I'll look into disabling secure boot again. I just recently did a fresh install of windows on that separate partition and haven't had a chance to do a full backup. Once I back it up I'll try disabling secure boot. 

Once secure boot is disabled, can I re-enable it once I get open razer running or is it something I would need to leave off?

lah7, 2018-10-29:
It would need to remain disabled in order to use the OpenRazer driver.

Secure boot simply prevents unsigned code from being loaded into the kernel. Since OpenRazer uses DKMS to compile the driver against your computer's installed kernel, you'd either need to sign it yourself or disable the "secure boot" feature.

tycruickshank, 2018-10-30:
Thanks so much for the help! For the time being I don't want to disable "secure boot" on this particular machine until I get a new external drive for backups.

z3ntu, 2018-11-01:
Closing this issue then.

closed 2018-11-01 by z3ntu