openrazer/openrazer #704: Razer Blade 15 (2018) Basic Model Support

aquahika, 2018-10-30:
__Product : Razer Blade 15 Keyboard__
__Model: (2018) Basic Model__

I found the Basic Model of Razer Blade 15(2018) has difference ID from 1532:0233.

My outputs of `lsusb | grep 1532`:
```
Bus 001 Device 004: ID 1532:023b Razer USA, Ltd 
```

Device Descriptor:
```
Device: ID 1532:023b Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x023b 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

The outputs of `lsusb -lah /dev/input/by-id/`:
```
total 0
drwxr-xr-x 2 root root 140 Oct 30 22:54 .
drwxr-xr-x 4 root root 540 Oct 30 23:23 ..
lrwxrwxrwx 1 root root   9 Oct 30 22:54 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event9
lrwxrwxrwx 1 root root   9 Oct 30 22:54 usb-Razer_Razer_Blade-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Oct 30 22:54 usb-Razer_Razer_Blade-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Oct 30 22:54 usb-Razer_Razer_Blade-if02-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Oct 30 22:54 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

zkarcher, 2018-12-05:
@aquahika I'd like to help. I have a Razer Blade 15 2018, currently dual-booting Windows 10 and Ubuntu 18.04.

I don't know how to sniff USB packets. If you tell me the steps, I'll do it! Thanks.

z3ntu, 2018-12-06:
@zkarcher See https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

KrisHedges, 2018-12-31:
I also have this model and can't get it working either. I've tried to follow along within the code as though it were like the other 2018 Models but it is too different and will require more than just adding the ID number.

On the razer site it states that the Advanced Model and Mercury White Model have "Per Key RGB Lighting" Where as this model has "Full Key Single-zone RGB Lighting".

I have all the same outputs as the top of this issue. Any suggestions on how I might proceed?

KrisHedges, 2019-01-02:
Ok I managed to get a stripped down version working with just static color. I'll look into it more tonight and determine which effects this thing can actually do in Synapse and try and get a more complete version going and then make a MR.

KrisHedges, 2019-01-03:
Here's a working version for the Razer Blade 2018 Base Model. See this pull request..
https://github.com/openrazer/openrazer/pull/750

AnnoyingTechnology, 2019-01-19:
This is great news, any idea when this will make it to the release state ? 
Thanks

KrisHedges, 2019-01-19:
My guess is it will merged in whenever the author goes through again it’s fully usable and no issues in the build ci. If you need support you can always just checkout the branch and install from there

AnnoyingTechnology, 2019-01-30:
I just built it, but nothing detected. Might have missed something. 

[image]

bendingunit15, 2019-02-11:
I can try to help with the RE if you still need a capture.

z3ntu, 2019-03-18:
Supported on master, so closing this issue.

closed 2019-03-18 by z3ntu