openrazer/openrazer #706: systemd-udevd[636]: Process 'razer_mount razerkbd 0003:1532:0221.0002' failed with exit code 1.

kierun, 2018-10-31:
My machine is running Fedora 28 (`Linux nightwatch 4.18.16-200.fc28.x86_64`) and before the last kernel update, the driver works fine -- if I boot that old kernel (4.18.14) the driver works fine.

`journalctl -b` highlight the error `systemd-udevd[636]: Process 'razer_mount razerkbd 0003:1532:0221.0002' failed with exit code 1.`

`sudo modprobe razerkbd` returns 0. I even tried rebuilding and installing the module via:
```
sudo dkms remove openrazer-driver/2.4.0 --all
sudo dkms install openrazer-driver/2.4.0
```

Any idea what could be happening?

What would you need from me?

z3ntu, 2018-10-31:
Does `sudo journalctl -b | grep -i razer_mount` print anything useful?

kierun, 2018-10-31:
This is what I get:

```
; sudo journalctl -b | grep -i razer_mount
1059:Oct 31 11:03:51 nightwatch razer_mount[659]: Driver razerkbd
1060:Oct 31 11:03:51 nightwatch razer_mount[661]: Driver razerkbd
1062:Oct 31 11:03:51 nightwatch razer_mount[662]: Driver razerkbd
1063:Oct 31 11:03:51 nightwatch razer_mount[664]: Device_ID 0003:1532:0221.0003
1064:Oct 31 11:03:51 nightwatch razer_mount[666]: Device_ID 0003:1532:0221.0001
1065:Oct 31 11:03:51 nightwatch razer_mount[671]: Modprobing razerkbd
1067:Oct 31 11:03:51 nightwatch systemd-udevd[626]: Process 'razer_mount razerkbd 0003:1532:0221.0001' failed with exit code 1.
1068:Oct 31 11:03:51 nightwatch razer_mount[668]: Device_ID 0003:1532:0221.0002
1069:Oct 31 11:03:51 nightwatch razer_mount[669]: Modprobing razerkbd
1070:Oct 31 11:03:52 nightwatch razer_mount[673]: Modprobing razerkbd
1071:Oct 31 11:03:52 nightwatch systemd-udevd[622]: Process 'razer_mount razerkbd 0003:1532:0221.0003' failed with exit code 1.
1072:Oct 31 11:03:52 nightwatch systemd-udevd[636]: Process 'razer_mount razerkbd 0003:1532:0221.0002' failed with exit code 1.
```

I am afraid it means little to me…

z3ntu, 2018-10-31:
`sudo modprobe razerfirefly; sleep 1; lsmod | grep razer` ?

kierun, 2018-10-31:
```
; sudo modprobe razerfirefly; sleep 1; lsmod | grep razer
2:razerfirefly           40960  0
3:razerkbd               69632  0
```

z3ntu, 2018-10-31:
No idea what's wrong. Have you tried to reboot?

You can also try running `sudo bash -x /usr/lib/udev/razer_mount razerkbd 0003:1532:0221.0002` in the terminal and see exactly where it fails. But looking at the code and your output, it fails at the modprobe command but as we can see, your kernel can load the modules just fine...

kierun, 2018-11-01:
> No idea what's wrong. 

☹ Thank you nonetheless for all your help.

> Have you tried to reboot?

No, I shall give that a go.

And here is the data from the command:

```
;  sudo bash -x /usr/lib/udev/razer_mount razerkbd 0003:1532:0221.0002
+ set -e
+ PATH=/sbin:/bin:/usr/sbin:/usr/bin
+ '[' -x /usr/bin/logger ']'
+ LOGGER=/usr/bin/logger
+ '[' -t 1 -a -z /usr/bin/logger ']'
+ '[' '!' -e /dev/log ']'
+ '[' -t 1 ']'
+ DRIVER=razerkbd
+ DEVICE_ID=0003:1532:0221.0002
+ mesg 'Driver razerkbd'
+ echo 'Driver razerkbd'
Driver razerkbd
+ /usr/bin/logger -t 'razer_mount[18221]' 'Driver razerkbd'
+ mesg 'Device_ID 0003:1532:0221.0002'
+ echo 'Device_ID 0003:1532:0221.0002'
Device_ID 0003:1532:0221.0002
+ /usr/bin/logger -t 'razer_mount[18221]' 'Device_ID 0003:1532:0221.0002'
+ '[' '!' -d /sys/bus/hid/drivers/razerkbd ']'
+ '[' -d /sys/bus/hid/drivers/hid-generic/0003:1532:0221.0002 ']'
+ '[' -d /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002 ']'
+ mesg 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ echo 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ /usr/bin/logger -t 'razer_mount[18221]' 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ chgrp -R plugdev /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ mesg 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ echo 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ /usr/bin/logger -t 'razer_mount[18221]' 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
```

kierun, 2018-11-01:
> No idea what's wrong. 

☹ Thank you nonetheless for all your help. It is much appreciated.

> Have you tried to reboot?

No, I shall give that a go. And that did the trick. I think the remove/reinstall command fixed it.

I thought I'd never see the day when rebooting a Linux machine would fix something⸮ ☺

And here is the data from the command:

```
;  sudo bash -x /usr/lib/udev/razer_mount razerkbd 0003:1532:0221.0002
+ set -e
+ PATH=/sbin:/bin:/usr/sbin:/usr/bin
+ '[' -x /usr/bin/logger ']'
+ LOGGER=/usr/bin/logger
+ '[' -t 1 -a -z /usr/bin/logger ']'
+ '[' '!' -e /dev/log ']'
+ '[' -t 1 ']'
+ DRIVER=razerkbd
+ DEVICE_ID=0003:1532:0221.0002
+ mesg 'Driver razerkbd'
+ echo 'Driver razerkbd'
Driver razerkbd
+ /usr/bin/logger -t 'razer_mount[18221]' 'Driver razerkbd'
+ mesg 'Device_ID 0003:1532:0221.0002'
+ echo 'Device_ID 0003:1532:0221.0002'
Device_ID 0003:1532:0221.0002
+ /usr/bin/logger -t 'razer_mount[18221]' 'Device_ID 0003:1532:0221.0002'
+ '[' '!' -d /sys/bus/hid/drivers/razerkbd ']'
+ '[' -d /sys/bus/hid/drivers/hid-generic/0003:1532:0221.0002 ']'
+ '[' -d /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002 ']'
+ mesg 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ echo 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ /usr/bin/logger -t 'razer_mount[18221]' 'Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ chgrp -R plugdev /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ mesg 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
+ echo 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/
+ /usr/bin/logger -t 'razer_mount[18221]' 'Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0221.0002/'
```

closed 2018-11-01 by kierun