openrazer/openrazer #707: Fedora 29 Repository Unavailable/Unlisted

Orochikaku, 2018-11-01:
On the [project page](https://openrazer.github.io/#download), under Fedora there are only repos for 27 and 28. The current version of Fedora is 29

z3ntu, 2018-11-01:
Thanks for the report. Even though the openSUSE Build Service currently has some kind of problem and everything is super slow, I have managed to enable builds for Fedora 29. 

Hopefully soon the builds will get published - check at https://download.opensuse.org/repositories/hardware:razer/ if there's a `Fedora_29` folder - and then follow the instructions on the website but adjust the URL for Fedora 29. :)

ctrl-alt-d, 2018-11-01:
Just upgrade to Fedora 29 using your instructions. All running. Thanks.

z3ntu, 2018-11-01:
Adjusted the website to list Fedora 29. Fixed!

closed 2018-11-01 by z3ntu