openrazer/openrazer #711: Release on PyPI

redaxmedia, 2018-11-06:
Hey folks,

could you please provide openrazer on PyPI? I would love to:

1. Get rid of the installation guides for each Linux distribution
1. Just add your library to [install_requires](https://github.com/redaxmedia/chroma-feedback/blob/master/setup.py#L31-L34) inside `setup.py`
1. Get rid of the ugly try/catch [ImportError](https://github.com/redaxmedia/chroma-feedback/blob/master/src/core.py#L10-L13) handling

z3ntu, 2018-11-08:
First @daniellawrence needs to his package on pypi called "openrazer" (see https://github.com/daniellawrence/openrazer-python/issues/1 and https://pypi.org/project/openrazer/) then I can get to uploading the pylib :)

redaxmedia, 2018-11-08:
I don't mind about the name for the moment. Can you release it under "openrazer-official" or something like that?

z3ntu, 2018-11-09:
Unfortunately we can't just upload the current pylib to pypi / have it installed standalone because `openrazer/client/macro.py` imports `openrazer_daemon.misc.macro` (i.e. depends on the daemon being installed)

You can try it yourself by creating a venv and installing https://test.pypi.org/project/openrazer/ (be sure to install `pip3 install dbus-python numpy` beforehand as the test instance contains outdated versions (that don't build for me))

redaxmedia, 2018-11-09:
Well, the daemon needs to be installed also via PyPI too, otherwise there is a dependency missing.
Can't you just add `dbus-python numpy` to the install_requires?

z3ntu, 2018-11-09:
I definitely won't put the daemon up onto pypi. Simple reason is that running the daemon without the kernel driver is completely unsupported and will probably just never find any device without any indication why. This is bad for both developers and end users.

Removing the (Python) dependency of the pylib on the daemon should be possible by basically copying `openrazer_daemon/misc/macro.py` and `openrazer_daemon/keyboard.py` into the pylib.

About the dbus-python numpy thing: Yes, it's in the install_requires but when using the command on the test pypi instance it will install dbus-python and numpy from the test instance which is either uploaded by random users or is outdated.

redaxmedia, 2018-11-09:
Hehe, sorry to bother you but why don't you upload the client, daemon and drivers then? :D

The idea behind this issue was to get rid of the manual steps and let the end users just install the software without apt-get install. A temporary installation does not improve the current one.

z3ntu, 2018-11-09:
The kernel driver isn't python... And requires dkms + kernel headers set up. I know that the current installation procedure is quite error prone because of several factors but there's no easy way to improve this with the current architecture.

redaxmedia, 2018-11-09:
What about a python wrapper that handles auto-installing the dkms + kernel? I don't know if this is a bad practice - just an idea.

z3ntu, 2018-11-09:
NO THANKS

z3ntu, 2018-11-09:
That sounds extremely bad and whatever. Best I can do for this issue is upload the pylib to pypi so you can get rid of the import error handling of the pylib and handle if the daemon is missing (which afaik the pylib will tell you)

z3ntu, 2018-11-09:
Don't worry, I've answered probably hundreds of questions where people had setup problems (missing kernel headers, not in plugdev group, wrongly installed, etc, etc) so if there was an easy way to fix those issues I would have done it already :)

redaxmedia, 2018-11-09:
So you are open to possible solutions? What about using something [apt-wrapper](https://pypi.org/project/apt-wrapper/) or [deb-pkg-tools](https://pypi.org/project/deb-pkg-tools/) for a openrazer-autoinstaller?

It does not look that bad, let's just be pragmatic.

z3ntu, 2018-11-09:
That would only work on Debian derivatives..

lu-zero, 2018-12-03:
moving the driver in userspace I guess it isn't a viable option, right?

z3ntu, 2018-12-03:
Even if the driver was just in user space, publishing everything on PyPI wouldn't work (well, technically you it would be possible but it wouldn't be a "daemon").
Also, see #382

z3ntu, 2018-12-03:
No...

redaxmedia, 2018-12-03:
A simple no does not solve issues^^ ... so let's close this one?

closed 2018-12-03 by z3ntu

redaxmedia, 2018-12-03:
Really? I hope you improve your soft skills some day, this is not the way how to deal with contributors.

tildearrow, 2018-12-04:
And I hope you understand you can't force him to release OpenRazer on PyPI. If you want it anyway, do it by yourself.

Plus eventually the Python bits will disappear (see razer_test)...

redaxmedia, 2018-12-04:
I have no problem that this request has been rejected... but how the communication is going from the beginning. Didn't aspect to have a chat with Linus Torvalds...

z3ntu, 2018-12-04:
@redaxmedia Feel free to do whatever you want with this software, write set-up scripts and whatever but I will not do that myself because it goes against what package managers are for and how software is being distributed on Linux.
As I wrote before, I am very well aware of the many problems that exist due to the existance of the kernel drivers, but as @tildearrow pointed out, I am working on a replacement for the current OpenRazer stack that is completely in userspace and really just requires one package being installed (no kernel headers, no plugdev group, no other nonsense) and having (in my opinion) many more advantages.

Writing distro-dependent setup scripts for non-Python software and publishing that on PyPI is just not the way things are done on Linux but if you don't accept that, I can't do more to change your view. I hope you understand (or at least accept) my position...

And replying to your deleted comment:
> but how the communication is going from the beginning

In the first 5 comments everything was about getting that working and the hurdles of doing so (also in the beginning I thought you were just about getting the python library up on pypi - because that made sense to me). In comment nr. 6 I explain why putting also the daemon up on PyPI is a bad idea and why it won't work. In the following comments I try to explain why you can't just upload C code meant to be built as a kernel module onto PyPI.

But again, this is open source. Do whatever you think is best, I can't stop you :slightly_smiling_face: