openrazer/openrazer #717: forum/mailing list and a question: keys-rebinding on blackwidow

ag88, 2018-11-23:
hi Devs,
thanks for putting together this great work, i just bought a Blackwidow Ultimate 2016 and stumbled into this. I've got some questions
1) forum / mailing list / google group / facebook group etc
other than this issues list is there a forum / mailing list / google group / facebook group where this (openrazer and related issues) is being discussed? I did some google searches but has not found any particular forums / mailing lists for discussions related to this.

2) keys rebinding
i made an attempt to install openrazer on my Opensuse linux system, but there are some missing dependencies as my distribution is too old. that aside, i've browsed the issues list and searched previous issues related to keys rebinding.

the closest i found are macros, there are some really good codes in the daemon that deals with the macros
https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/macro.py
https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/misc/key_event_management.py

however, am i correct to say that macros basically records the key sequences and are stored on the host / pc? that would be quite similar to keys rebinding say in KDE.

i'm actually hoping to redefine some keys on the keyboard itself, in particular i've had an old keyboard that has a 'power' key, pressing that key surprisingly sends a powerdown keycode in both Linux and Windows environment. this is different from macros as the binding needs to be done within the  keyboard rather than in the OS as the macros would be 'lost' when i reboot into windows.

is keys rebinding possible on Razer (Blackwidow (Ultimate)) keyboards?

ag88, 2018-11-23:
thanks! given that things has evolved till date and that Blackwidow Ultimate 
https://www.flickr.com/photos/mondalaci/5647291879/in/album-72157626562246038/
apparently uses a Freescale/NXP MC9S08JM16 
https://cache.nxp.com/docs/en/data-sheet/MC9S08JM16.pdf
which apparently has only 16k flash and 1k sram, the firmware i'd guess would need all that resources to handle usb and a static keymap
it would seem that the functionality of storing remapped kays may not be available after all given the very limited resources on the mcu,

z3ntu, 2019-03-23:
Closing for now.

closed 2019-03-23 by z3ntu