openrazer/openrazer #719: Add support for razer Goliathus Extended Chome 1532:0C02

Mammothskier, 2018-11-24:
[ls.txt](https://github.com/openrazer/openrazer/files/2612610/ls.txt)
[lsusb.txt](https://github.com/openrazer/openrazer/files/2612611/lsusb.txt)

fabianishere, 2019-01-12:
Hi,

I made some [traces](https://github.com/openrazer/openrazer/files/2751956/traces.zip) for the Razer Goliathus Extended Chroma.
I have found that, similar to the Firefly, it communicates at interface number 0x0 (instead of 0x2).

z3ntu, 2019-01-15:
@fabianishere Please could you redo the traces and make sure to save the captured data as `.pcapng` and not as `.pcap` ?

fabianishere, 2019-01-15:
@z3ntu Will do! Any specific effects you want me to capture? I already found that Synapse uses software-based lighting similar to #617, but still supports some of the built-in effects (e.g. spectrum)

fabianishere, 2019-01-29:
Apologies for the delay, I did not have access to it last week.

This [archive](https://github.com/openrazer/openrazer/files/2808141/razer-goliathus-chroma-extended-dissection.pcapng.zip) contains a small trace of the Goliathus Extended. I have annotated some packets for identification.

kevindurb, 2019-02-12:
Anything i can do to help? I have a Golithius extended too and im dual booting windows and fedora

gaspoza, 2019-03-18:
@fabianishere 
Hello
Based on your comment from Jan 12, I understood that you have made some change in order to get working the Goliathus Extended Chroma.
Can you please explain these steps? Just want to be able to set a static color...
Any update on this issue? Date of the next release?

fabianishere, 2019-03-18:
@gaspoza 
Hi,
I did this by manually sending HID packets to the device similar to OpenRazer. I did start on a driver implementation, but haven't been able to finish it yet.

You can take a peek here: https://github.com/fabianishere/openrazer/tree/feature_goliathus_0c02

intervigilium, 2019-03-23:
Goliathus is using 0x00 for the LED ID field. If you switch your razer_chroma_extended_matrix_effect function calls to pass 0x00 instead of 0x1 all the effects work. Reactive is not a very useful effect for the Goliathus though, since it pulses based on input, and a mousepad has no input.

fabianishere, 2019-03-23:
@intervigilium That’s right! I just copied over de source from the Firefly and tried to get static effects working first. 😄

I will check on it again this weekend to see if I can finish it.

z3ntu, 2019-03-28:
Support was merged in #796

closed 2019-03-28 by z3ntu