openrazer/openrazer #722: Dkms package not found Manjaro 4.9.140-1

SirKillo0o, 2018-11-27:
Hi,

It's my first time posting in this page so i apologize if I'm braking any rules.

Going down the rabbit hole I'm using a razer Ouroboros and UIs using the openrazer API can't detect the device (my guess is that the daemon can't bind the device)

Currently I'm trying to install the dkms package but I get this:
$ sudo dkms install openrazer-driver/1.0.0
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-1.0.0 does not exist.

I'm a bit of a noob to this so I would be happy for any help.

thanks in advance

SirKillo0o, 2018-11-27:
You were spot on with the headers and now that the driver is installed everything lit up and works.

Thought could you please tell me where you found the version which had to be installed?

Thanks in advance

z3ntu, 2018-11-27:
https://github.com/openrazer/openrazer/releases
2.4.0 is the latest one.

closed 2018-11-27 by z3ntu