openrazer/openrazer #724: Add support for razer Nari 1532:051d

vktrbhm, 2018-11-30:
Please add support for razer Nari 1532:051d

z3ntu, 2018-11-30:
* Please post the output of
`lsusb -d 1532:051d -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

cra0zy, 2019-01-01:
On a related side note, Nari does not work as an audio device out of the box properly, so this is probably gonna be a way more challenging to implement than just checking and adding chroma controller code.

z3ntu, 2019-01-15:
This project is basically code that calls APIs on the device to change the LEDs & related behavior. It is not for adding basic kernel support for it.

sjfricke, 2019-01-19:
My Nari Ultimates worked out of the box for my Ubuntu 18.04 machine, both wired and wirelessly

cra0zy, 2019-01-19:
Mine does work wireless, but only mono sound output/chat output gets detected from linux.

Same as these guys: https://superuser.com/questions/1390303/razer-nari-headphones-do-not-work-wirelessly-on-ubuntu-18

sjfricke, 2019-01-19:
I just realized there are the Nari and the Nari Ultimates (which I have)... I will have to do some more digging into this, I would be very willing to help get my headphones to have *some* linux support

DarkSaber203, 2019-05-28:
Mine are the same as @cra0zy only mono on wireless. like i don't fully mind but listening to music or movies drives me nuts. i switched over from windows and so far this is my only razer issue.
[image]

taylorcoffelt, 2019-05-29:
I have the Razer Nari (1532:051c) [Not the Ultimate or Essential](https://www.razer.com/gaming-audio/razer-nari)

<details>
    <summary>
Output of lsusb -d 1532:051c -v
</summary> <p>

``` Bus 001 Device 006: ID 1532:051c Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051c 
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          445
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 Razer Nari - Chat
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength           70
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
        bSourceID               6
        bControlSize            1
[... 365 lines cut ...]
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
</p>
</details>
<details>
<summary>Output of ls -lah /dev/input/by-id/</summary><p>

```
lrwxrwxrwx 1 root root  10 May 28 20:06 usb-Razer_Razer_Nari-event-if05 -> ../event10
```
</p>
</details>

Model Number of Headset: `RC30-026801`
Model Number of USB Dongle: `RC30-0267`

I'll see about getting screenshots of Razer Synapse and Packet caps soon

avengerweb, 2019-06-16:
Connected to USB:
**1532:051d**
`
lsusb -d  1532:051d -v

Bus 005 Device 008: ID 1532:051d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051d 
  bcdDevice            1.00
  iManufacturer           2 Razer
  iProduct                3 Razer Nari
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              1 Hid Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     101
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

`

**Wireless: 1532:051c**
`
lsusb -d  1532:051c -v

[... 413 lines cut ...]
      bInterfaceProtocol      0 None
      iInterface              3 Hid Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
`

ItchieR, 2019-07-31:
Hello, I have newly purchased a Razer Nari and want to have support for it, I may also can help to get the needed information. Is there a possible way to get this device working in the near future? Is there anything I can help with my linux system and the nari device?

DarkSaber203, 2019-08-17:
> * Please post the output of
>   `lsusb -d 1532:051d -v`
>   and
>   `ls -lah /dev/input/by-id/`
> * Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we [...]

i see someone posted the info for the headset, is their any word on getting it supported.

imustafin, 2020-02-26:
> I have the Razer Nari (1532:051c) [Not the Ultimate or Essential](https://www.razer.com/gaming-audio/razer-nari)
> Output of lsusb -d 1532:051c -v
> Output of ls -lah /dev/input/by-id/
> 
> ```
> lrwxrwxrwx 1 root root  10 May 28 20:06 usb-Razer_Razer_Nari-event-if05 -> ../event10
> ```
> 
> Model Number of Headset: `RC30-026801`
> Model Number of USB Dongle: `RC30-0267`
> 
> I'll see about getting screenshots of Razer Synapse and Packet caps soon

I have this device. I can collect packet capture if needed.

I also have this problem when the headset has only mono mode on wireless:
> Mine are the same as @cra0zy only mono on wireless. like i don't fully mind but listening to music or movies drives me nuts. i switched over from windows and so far this is my only razer issue.
> [image]

From manual I understand that on Windows the wireless dongle should give two audio devices: one for games, one for chat. Probably the one which is recognized on Linux is mono, while the other can be stereo. Recognizing the stereo device can be a great improvement to the usability of this headset.

I will check on Windows if this is really true that one of the devices is mono.

imustafin, 2020-02-27:
When collecting the pcap, what actions should I do?

Also, would it be possible to at least solve the problem with Mono/Stereo sound output at least, before working on other things? Are there any other devices which had similar issue to see how the problem was solved there?

imustafin, 2020-02-27:
Nevermind the sound problems. There is a solution for Pulseaudio https://github.com/imustafin/razer-nari-pulseaudio-profile.

Later I will bring pcaps and screenshots.

avengerweb, 2020-02-29:
> Nevermind the sound problems. There is a solution for Pulseaudio https://github.com/imustafin/razer-nari-pulseaudio-profile.
> 
> Later I will bring pcaps and screenshots.

Now input device is missing, how to fix it?

imustafin, 2020-03-01:
@avengerweb maybe here is not the best place to discuss it. Open an issue in that repo with outputs of `pacmd list-sources` before and after applying the fix.

harry-cpp, 2020-03-01:
@imustafin You need to enable issues on your fork.

imustafin, 2020-03-01:
@harry-cpp oops thanks. I've enabled them.

imustafin, 2020-06-11:
I've taken a look at the packets and I am a bit confused.

I have Windows 10 on a VM with Synapse 3 communicating with my 1532:051c Nari.

I think I've managed to identify the commands to set brightness and color, however the protocol seems very different from what was described in wiki https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol and from drivers of other devices.

I think that the protocol is really different case because it seems that Nari doesn't store it's mode. If I set any mode and then disconnect Nari, then it goes back to its default Static Green mode.

Are there any "Synapse 3 era" devices supported by openrazer? I'll experiment a bit more and make a draft PR with my findings.

Attaching packets, here I just play with brightness.
[brightness_0_100.pcapng.zip](https://github.com/openrazer/openrazer/files/4764208/brightness_0_100.pcapng.zip)

j3luehawk097, 2020-07-08:
I own a normal Razer Nari and I am sick and tired of having to deal with Synapse.... Can you please let me know how I can help to move this forward! I am willing to do anything it takes to get rid of it, please!

imustafin, 2020-07-08:
@j3luehawk097 what is your problem? I also own a normal Nari.

If you have problems with sound channel being recognized as mono, then you can check out https://github.com/imustafin/razer-nari-pulseaudio-profile. Feel free to open an issue in that repo if there are any questions.

Or do you want some other features provided by Synapse?

j3luehawk097, 2020-07-09:
@imustafin It just does not work wirelessly on PopOs! Also, I did not see it in the supported devices list. Regarding Synapse, nothing, just wish to have this headset work and have some of the control over the devices available.

carlosgvaso, 2020-07-13:
I have the Razer Nari Essential headset, and I gathered this data on Ubuntu 20.04 LTS.

* Headset: Razer Nari Essential:
    * Model number: RC30-026901
    * Product number: RC30-02690100-0000 / RZ04-02690100-R3U1
* USB Wireless Transceiver:
    * Model number: RC30-026902
    * Product number: N/A

Let me know if there is more info needed. I can help get packet captures if needed too.

Using the USB Wireless Transceiver, the headset shows up on the settings with mic and headphones options, and both mic and headphones work well out of the box. Here is the info requested using the transceiver:

<details>
  <summary>`$ lsusb -d 1532: -v`</summary>

```
$ lsusb -d 1532: -v

Couldn't open device, some information will be missing

Bus 001 Device 011: ID 1532:051e Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051e 
  bcdDevice            1.00
  iManufacturer           3 
  iProduct                4 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0144
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0047
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
[... 364 lines cut ...]

```
</details>

<details>
  <summary>`$ ls -lah /dev/input/by-id/`</summary>

```
$ ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 220 Jul 13 09:08 .
drwxr-xr-x 4 root root 580 Jul 13 09:08 ..
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-046d_C922_Pro_Stream_Webcam_05DC39DF-event-if00 -> ../event9
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-if02 -> ../event7
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if01-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Microsoft_Microsoft®_2.4GHz_Transceiver_v9.0-if02-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Valve_Software_Steam_Controller-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Jul 13 03:37 usb-Valve_Software_Steam_Controller-mouse -> ../mouse1

```
</details>

TermoZour, 2020-10-06:
According to @imustafin 's pcaps as well as my own captures, I have found the 3 values responsible for controlling the RGB on these. If I am not mistaken, byte 73, 74, 75 are R, G, B respectively. How can we manipulate these bytes for further testing? The Wiki kind of stops at "capturing packets"

TermoZour, 2020-10-06:
Here I have captured some packets with static color.
1. color set to FF 00 00
2. color set to 00 FF 00
3. color set to 00 00 FF
4. color set to FF FF FF

As you can see, bytes 73, 74, 75 are what change.
[rgb-sequence-r-g-b-white.pcapng.gz](https://github.com/openrazer/openrazer/files/5335326/rgb-sequence-r-g-b-white.pcapng.gz)

pieromarini, 2020-11-16:
Hello, I own a Razer Nari Ultimate **(1532:0051a)**. I know this issue is for the Normal version of the Nari, but since the protocols are the same, the information should be useful anyways.
A few months back I was checking this issue and came up with a simple python program to modify, successfully, the colors and create semi-decent transitions between them.
The script can be found [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513). It uses [PyUSB](https://github.com/pyusb/pyusb) to communicate with the device. The script is heavily commented.
From my RE, the protocol is completely different as @imustafin mentioned. There are no more "modes" to set. The device just receives a stream of packets indicating the color it has to set. After that, it stays with that color until it is turned off / disconnected (it then goes to the default green with "Static mode"). 

In the provided script [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513#file-nari_ultimate_testing-py-L30), I use a simple lerp to create a transition between two colors and just send them to the device with a small delay in-between [here](https://gist.github.com/pieromarini/4ec31252b4de76f2181a0467eccfa513#file-nari_ultimate_testing-py-L60). Following this logic, and from what I found using packet captures **(found at the bottom)** from a Windows 10 VM, we just need to send these series of packets to create the desired transitions. 

For the case of adjusting brightness, the logic is similar, although it would only be 1 packet sent. For the Nari Ultimate, you can also set On/Off the "vibration" (Haptic Feedback) feature from synapse. This was also just a packet but I'm dumb and didn't save the format it has.

I think this is enough information to build a driver. I don't have much experience writing kernel drivers and I haven't had any spare time to take a look at it recently, but I hope someone can take this information and write it.

<details>
    <summary>Output of <code>lsusb -d 1532:0051a -v</code> for reference</summary>

```
Bus 001 Device 013: ID 1532:051a Razer USA, Ltd Nari Ultimate
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051a Nari Ultimate
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari Ultimate
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0
      iInterface              1 Razer Nari Ultimate - Chat
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0
[... 362 lines cut ...]
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```
</details>

[Packet Captures (includes static, breathing and spectrum RGB modes)](https://github.com/openrazer/openrazer/files/5543470/pcaps.zip)

z3ntu, 2020-11-17:
@pieromarini Could you please copy your comment to #974 so we don't mix up the devices? The Nari (& Nari Ultimate) seem to use a different protocol than the rest of the Razer devices for whatever reason so I can't really help with that device unfortunately.

Yozaru456, 2020-11-18:
Hello,
I own a Razer Nari  . I am on PoP Os 20.04 and my sound keeps getting lower and then  jumps back . All the damn time . Is this issue known ?

TermoZour, 2020-11-18:
> Hello,
> I own a Razer Nari  . I am on PoP Os 20.04 and my sound keeps getting lower and then  jumps back . All the damn time . Is this issue known ?

I am currently using Nari on Pop!OS 20.04 and do not have this issue. I'm using Pulseaudio btw.

free-vbucks, 2020-11-25:
Pcap with 7.1 surround sound and stereo surround sound put into the headphones
Using a chroma rainbow "wheel" effect for the coloring

[nari.pcapng.zip](https://github.com/openrazer/openrazer/files/5599223/nari.pcapng.zip)

free-vbucks, 2020-12-08:
This needs to be marked as not needing PCAP

TermoZour, 2020-12-10:
I filtered it so only Razer Nari will show up. (I also have a few entries for other Razer gear but those are irrelevant right now)
```
ls -lah /dev/input/by-id

lrwxrwxrwx 1 root root  10 Dec 10 11:41 usb-Razer_Razer_Nari-event-if05 -> ../event15
```

mercedes7935, 2021-03-12:
I am sorry for maybe necroing this but I currently thinking about buying the razer nari ultimate but i ask myself several questions like does it save change if I make them in windows and use the headset on Linux or another PC and Linux,..., is there anything bad at the nari ultimate like a low durability or something other which is awful? Is there a way to configure the headset under Linux/ for Linux with every / most of the thing in synapse(lightning not including i don't care about lightning green is nice)

TermoZour, 2021-03-12:
> I am sorry for maybe necroing this but I currently thinking about buying the razer nari ultimate but i ask myself several questions like does it save change if I make them in windows and use the [...]

I personally didn't test the audio settings to see if they are still there on Linux or if they're Windows specific (because of the windows driver). 

THX surround doesn't "officially work". You can have virtual surround set up in linux for any headphones in PulseAudio (they sound amazing), headphones don't need the "THX Certified" logo.

What I know for sure remains stored is the haptic feedback. If I set it in Windows, it sticks no matter where or how I use the headphones (wireless or wired, OS agnostic).

In terms of quality, they seem better built than the regular Nari (I have both models). The volume sliders (game/chat as well as master volume) seem better built. 

RGB you can set with a Python script. I'd assume you can also set the haptic feedback via the same method but I'm not sure. You can also fire up a basic Windows VM and tweak settings from there.

mercedes7935, 2021-03-12:
> > I am sorry for maybe necroing this but I currently thinking about buying the razer nari ultimate but i ask myself several questions like does it save change if I make them in windows and use the headset on Linux or another PC and Linux,..., is there anything bad at the nari ultimate like a low durability or something other which is awful? Is there a way to configure the headset under Linux/ for Linux with every / most of the thing in synapse(lightning not including i don't care about lightning green is nice)
> 
> I personally didn't test the audio settings to see if they still have any effect on Linux or if they're Windows specific (because of the windows driver).
> 
> THX surround doesn't "officially work". You can have virtual surround set up in linux for any headphones in PulseAudio (they sound amazing), headphones don't need the "THX Certified" logo.
> 
> What I know for sure remains stored is the haptic feedback. If I set it in Windows, it sticks no matter where or how I use the headphones (wireless or wired, OS agnostic).
> 
> In terms of quality, they seem better built than the regular Nari (I have both models). The volume sliders (game/chat as well as master volume) seem better built.
> 
> RGB you can set with a Python script. I'd assume you can also set the haptic feedback via the same method but I'm not sure. You can also fire up a basic Windows VM and tweak settings from there.

Ok thanks this helps much! how long do you own 1-2 years right? And thanks for even replying nearly instantly!

TermoZour, 2021-03-12:
> Ok thanks this helps much! how long do you own 1-2 years right? And thanks for even replying nearly instantly!

I've had the Nari Ultimate for about 4 months. The Nari for about 5 months.

Since I started playing some more multiplayer games, I kept switching to Windows more often for the "gaming". Otherwise, the headphones are working perfectly fine on Linux. Just don't forget to use the PulseAudio profile for them. Otherwise Linux will only recognize the mono (chat) channel.

mercedes7935, 2021-03-12:
> I've had the Nari Ultimate for about 4 months. The Nari for about 5 months.
>
> Since I started playing some more multiplayer games, I kept switching to Windows more often for the "gaming". Otherwise, [...]

Ok thanks I searched how I can set up such a profile need for THX but didn't found a guide can you explain to me how I can do it?

TermoZour, 2021-03-12:
@mercedes7935 I've followed this guide: https://forum.endeavouros.com/t/howto-setting-up-virtual-surround-sound-for-headphones/6889

bushblade, 2021-12-06:
I see that this issue is closed though there the Nari doesn't seem to be supported? What's the situation currently?

minecrawler, 2021-12-06:
@bushblade This issue is still open (take a look at the status left of the title)!

The status is: Pending until someone implements it.

Agorass, 2023-12-06:
Good evening is this "Device Support" still active? Cause all other Devices now support except my 0x051c Nari (Wireless). 

`Bus 001 Device 007: ID 1532:051c Razer USA, Ltd Nari (Wireless)
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051c Nari (Wireless)
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
        bSourceID               6
        bControlSize            1
        bmaControls(0)       0x03
          Mute Control
          Volume Control
        bmaControls(1)       0x00
[... 368 lines cut ...]
`

`drwxr-xr-x 2 root root 440 Dec  6 18:55 .
drwxr-xr-x 4 root root 920 Dec  6 18:55 ..
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Base_Station_Chroma-event-mouse -> ../event17
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Base_Station_Chroma-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Base_Station_Chroma-if02-event-kbd -> ../event20
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_Base_Station_Chroma-mouse -> ../mouse4
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Basilisk_Ultimate_Dongle-event-mouse -> ../event18
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Basilisk_Ultimate_Dongle-if02-event-kbd -> ../event13
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_Basilisk_Ultimate_Dongle-mouse -> ../mouse1
lrwxrwxrwx 1 root root  11 Dec  6 18:55 usb-Razer_Razer_BlackWidow_V4_Pro-event-if04 -> ../event268
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_BlackWidow_V4_Pro-event-kbd -> ../event3
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_BlackWidow_V4_Pro-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_BlackWidow_V4_Pro-if02-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_BlackWidow_V4_Pro-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root  11 Dec  6 18:55 usb-RAZER_Razer_Firefly_V2-event-mouse -> ../event267
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-RAZER_Razer_Firefly_V2-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Mouse_Dock-event-mouse -> ../event16
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Mouse_Dock-if01-event-kbd -> ../event15
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Mouse_Dock-if02-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 Dec  6 18:55 usb-Razer_Razer_Mouse_Dock-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 Dec  6 18:55 usb-Razer_Razer_Nari-event-if05 -> ../event29
`