openrazer/openrazer #725: OpenRazer uploads effects too slowly

tildearrow, 2018-12-01:
So I've been writing an application using libopenrazer to do some custom effects, but I just found a problem: the OpenRazer driver (and especially the daemon) is too slow to upload the matrix, and when I terminate my application in my 3-device setup it has to "catch up", which means it's not running full-speed...

1-and-half years ago I found out a certain OpenRazer (then razer-drivers) slowed down the uploading commands on keyboards for some reason, but I wasn't able to bisect the issue, and it wasn't that important to me in that time since my older application talked directly to the driver and as such I could thread.

Steps to reproduce:
1. Run the following command:
```
echo "AAAV////////////////////////////////////////////////////////////////////////////////////////AQAV////////////////////////////////////////////////////////////////////////////////////////AgAV////////////////////////////////////////////////////////////////////////////////////////AwAV////////////////////////////////////////////////////////////////////////////////////////BAAV////////////////////////////////////////////////////////////////////////////////////////BQAV////////////////////////////////////////////////////////////////////////////////////////" | base64 -d > upBench
{ while true; do cat upBench > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/matrix_custom_frame; echo "1" > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/matrix_effect_custom; echo -n .; done; } | pv -r > /dev/null
```

Speed in razer-drivers v1.0.0:
```
echo "AP///////////////////////////////////////////////////////////////////////////////////////wH///////////////////////////////////////////////////////////////////////////////////////8C////////////////////////////////////////////////////////////////////////////////////////A////////////////////////////////////////////////////////////////////////////////////////wT///////////////////////////////////////////////////////////////////////////////////////8F////////////////////////////////////////////////////////////////////////////////////////" | base64 -d >upBenchOld # had to be modified for older version
$ { while true; do cat upBenchOld > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/set_key_row; echo "1" > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/mode_custom; echo -n .; done; } | pv -r > /dev/null
[88.4 B/s]
```

Speed in OpenRazer v2.4.0:
```
$ { while true; do cat upBench > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/matrix_custom_frame; echo "1" > /sys/bus/hid/drivers/razerkbd/0003\:1532\:0216.000C/matrix_effect_custom; echo -n .; done; } | pv -r > /dev/null
[55.1 B/s]
```

As you can see, OpenRazer is much slower. What could be causing this?

z3ntu, 2018-12-01:
No idea.

tildearrow, 2018-12-01:
I bisected the issue to this commit:
0a67aedd

(the 2 commits before that one are insanely faster, but not sure whether that's because it was incomplete)

tildearrow, 2018-12-02:
Good news: I just found out the issue.
There is an unnecessary sleep in razer_get_usb_response (marked with "THIS SLEEP IS NOT NEEDED"):

```C
int razer_get_usb_response(struct usb_device *usb_dev, uint report_index, struct razer_report* request_report, uint response_index, struct razer_report* response_report, ulong wait_min, ulong wait_max)
{
    uint request = HID_REQ_GET_REPORT; // 0x01
    uint request_type = USB_TYPE_CLASS | USB_RECIP_INTERFACE | USB_DIR_IN; // 0xA1
    uint value = 0x300;

    uint size = RAZER_USB_REPORT_LEN; // 0x90
    int len;
    int retval;
    int result = 0;
    char *buf;

    buf = kzalloc(sizeof(struct razer_report), GFP_KERNEL);
    if (buf == NULL)
        return -ENOMEM;

    // Send the request to the device.
    // TODO look to see if index needs to be different for the request and the response
    retval = razer_send_control_msg(usb_dev, request_report, report_index, wait_min, wait_max);

    // Now ask for response
    len = usb_control_msg(usb_dev, usb_rcvctrlpipe(usb_dev, 0),
                          request,         // Request
                          request_type,    // RequestType
                          value,           // Value
                          response_index,  // Index
                          buf,             // Data
                          size,
                          USB_CTRL_SET_TIMEOUT);

    // > > > THIS SLEEP IS NOT NEEDED < < <
    usleep_range(wait_min, wait_max);

    memcpy(response_report, buf, sizeof(struct razer_report));
    kfree(buf);

    // Error if report is wrong length
    if(len != 90) {
        printk(KERN_WARNING "razer driver: Invalid USB response. USB Report length: %d\n", len);
        result = 1;
    }

    return result;
}
```

After removing the sleep, the speed greatly increases:
```
[90.7 B/s]
```

And if you are worried if this may increase the error rate when fetching the device's serial or firmware version, don't worry, because I have tested with a little shell command and it seems to not have increased.

However, this may cause issues in other devices, so I am not sure whether the sleep removal can be applied.

z3ntu, 2018-12-03:
Send a PR please.

tildearrow, 2018-12-03:
Proceeded on doing so.

closed 2018-12-03 by tildearrow