openrazer/openrazer #728: [Debian-Buster][Razer Naga Hex][1532:0036][RZ01-0075] Mouse not recognized by Xwindows

TriMoon, 2018-12-05:
Hi, im new to using openrazer, that said:

My mouse was working correctly before i installed openrazer, but after it the mouse is not being recognized by XWindows as it does not respond to my mouse moves anymore.
**What is needed to get it recognized without uninstalling openrazer?**

Below are some info i think is relevant and i saw being asked in other issues:
* **OS**:
  * Debian Buster
* My Naga Hex:
  * Partnumber RZ01-00750100
  * is a GREEN colored version like the one shown on https://github.com/openrazer/openrazer/issues/149#issuecomment-271787118

```console
~/ lsusb|grep -i mouse
Bus 002 Device 002: ID 062a:4101 Creative Labs Wireless Keyboard/Mouse
Bus 002 Device 008: ID 1532:0036 Razer USA, Ltd RZ01-0075, Gaming Mouse [Naga Hex]
```

```console
~/ sudo journalctl -b | grep -i razer_mount
Dec 05 11:49:39 debian razer_mount[710]: Driver razermouse
Dec 05 11:49:39 debian razer_mount[711]: Driver razermouse
Dec 05 11:49:39 debian razer_mount[726]: Device_ID 0003:1532:0036.0004
Dec 05 11:49:39 debian razer_mount[727]: Device_ID 0003:1532:0036.0005
Dec 05 11:49:39 debian razer_mount[728]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0036.0004/
Dec 05 11:49:39 debian razer_mount[729]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0036.0005/
Dec 05 11:49:39 debian razer_mount[745]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0036.0005/
Dec 05 11:49:39 debian razer_mount[746]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0036.0004/
```

```console
grep "2018-12-05 16:12" ~/.local/share/openrazer/logs/razer.log
2018-12-05 16:12:44 | razer                          | INFO     | Initialising Daemon (v2.4.0). Pid: 15792
2018-12-05 16:12:44 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerCynosaChroma ----------------------- (1532:022A)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-12-05 16:12:44 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
[... 170 lines cut ...]
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               8
```

```console
~/ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 240 Dec  5 15:50 .
drwxr-xr-x 4 root root 660 Dec  5 15:50 ..
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-Elite_Silicon_USB_Audio_Device-event-if02 -> ../event7
lrwxrwxrwx 1 root root  10 Dec  5 11:49 usb-Ideazon_Merc_Stealth-event-if01 -> ../event15
lrwxrwxrwx 1 root root  10 Dec  5 11:49 usb-Ideazon_Merc_Stealth-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-MOSART_Semi._2.4G_Keyboard_Mouse-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-MOSART_Semi._2.4G_Keyboard_Mouse-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-MOSART_Semi._2.4G_Keyboard_Mouse-if01-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-MOSART_Semi._2.4G_Keyboard_Mouse-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-Razer_Razer_Naga_Hex-event-mouse -> ../event8
lrwxrwxrwx 1 root root  10 Dec  5 11:49 usb-Razer_Razer_Naga_Hex-if01-event-kbd -> ../event23
lrwxrwxrwx 1 root root   9 Dec  5 11:49 usb-Razer_Razer_Naga_Hex-mouse -> ../mouse1
```

z3ntu, 2018-12-06:
We do not have any support for key remapping currently.

If you have more problems, just reply in this thread.

closed 2018-12-06 by z3ntu

TriMoon, 2022-04-01:
@z3ntu I was just browsing my old posts and came across this again...
I noticed my mouse is still being referred to as `Razer Naga Hex (Red) | 1532:0036` while mine is green :wink: 
Maybe time to change/update that? :thinking:

z3ntu, 2022-04-01:
@TriMoon If you can think of a better name, feel free to suggest. I don't have any info on what colors this device came in so I cannot really do much here... And I think we're trying to avoid multiple devices with identical names and we already have a "Razer Naga Hex"

TriMoon, 2022-04-01:
@z3ntu 
I don't know either, so many "Colored" ?
I have no idea about what color is refered to with "Razer Naga Hex"...
But a Green device refered to as "Red" is wrong, i assume you agree... :wink: