openrazer/openrazer #729: Razer Tartarus device not recognised and driver not found

Aeodwrath, 2018-12-08:
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.12.14-lp150.12.25-default

lsusb | grep "1532"
Bus 001 Device 003: ID 1532:0208 Razer USA, Ltd 

sudo journalctl -b | grep -i razer_mount
Dec 08 14:17:03 USB3 razer_mount[1040]: Driver razerkbd
Dec 08 14:17:03 USB3 razer_mount[1042]: Device_ID 0003:1532:0208.0004
Dec 08 14:17:03 USB3 razer_mount[1045]: Modprobing razerkbd
Dec 08 14:17:03 USB3 razer_mount[1044]: Driver razerkbd
Dec 08 14:17:03 USB3 razer_mount[1050]: Device_ID 0003:1532:0208.0002
Dec 08 14:17:03 USB3 systemd-udevd[668]: Process 'razer_mount razerkbd 0003:1532:0208.0004' failed with exit code 1.                                                                                          
Dec 08 14:17:03 USB3 razer_mount[1051]: Modprobing razerkbd
Dec 08 14:17:03 USB3 razer_mount[1043]: Driver razerkbd
Dec 08 14:17:03 USB3 systemd-udevd[676]: Process 'razer_mount razerkbd 0003:1532:0208.0002' failed with exit code 1.                                                                                          
Dec 08 14:17:03 USB3 razer_mount[1056]: Device_ID 0003:1532:0208.0003
Dec 08 14:17:03 USB3 razer_mount[1059]: Modprobing razerkbd
Dec 08 14:17:03 USB3 systemd-udevd[687]: Process 'razer_mount razerkbd 0003:1532:0208.0003' failed with exit code 1. 

Running openSuse Leap 15 and installed via Yast.  Razer Genie has dependancy issues.
The only modules that runs are the polychromatic tray applet and controller. Controller says "Device not recognised"

z3ntu, 2018-12-08:
What's the output of the command `sudo dkms install openrazer-driver/2.4.0` ?

Aeodwrath, 2018-12-08:
That was interesting..
reating symlink /var/lib/dkms/openrazer-driver/2.4.0/source ->
                 /usr/src/openrazer-driver-2.4.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
KERNELDIR=/lib/modules/4.12.14-lp150.12.25-default/build make driver.....
cleaning build area...

DKMS: build completed.

razerkbd.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/

razermouse.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/

razerfirefly.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/

razerkraken.ko:
Running module version sanity check.
 - Original module                                                                                     
   - No original module exists within this kernel                                                      
 - Installation                                                                                        
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/                                   
                                                                                                       
razermug.ko:                                                                                           
Running module version sanity check.                                                                   
 - Original module                                                                                     
   - No original module exists within this kernel                                                      
 - Installation                                                                                        
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/                                   
                                                                                                       
razercore.ko:                                                                                          
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.12.14-lp150.12.25-default/updates/
Adding any weak-modules
depmod: ERROR: fstatat(5, nvidia.ko): No such file or directory
depmod: ERROR: fstatat(5, nvidia-drm.ko): No such file or directory
depmod: ERROR: fstatat(5, nvidia-modeset.ko): No such file or directory
depmod: ERROR: fstatat(5, nvidia-uvm.ko): No such file or directory

depmod...

DKMS: install completed.

z3ntu, 2018-12-08:
Looks like a problem with the nvidia modules. But you can try to reboot and maybe it'll work now.

Aeodwrath, 2018-12-08:
Ok Thanks for that, the device is recognised and the RazerGenie runs.  

However the Razer Configuration does not - does this work with the Tartarus - does it have key mapping support?

z3ntu, 2018-12-08:
No, key remapping is not supported. There are some other standalone tools that do that though.
Great, that it works now!

closed 2018-12-08 by z3ntu