openrazer/openrazer #736: Firefly RGB shuts off after a minute

miqued, 2018-12-17:
Using Ubuntu 18.04, default DE. The software, with People as the front end, correctly detected my Deathadder Elite and Firefly, and there are no issues using the software.

However, after about a minute, the lights on the Firefly shut off completely. Changing any of the settings for it or restarting the daemon will cause it to light up again, but only for another minute before the process needs to be repeated.

This continues across reboots.

miqued, 2018-12-19:
I'm not sure if this was seen or not or if an update was pushed out to fix this, but I am no longer having this issue. Thanks for making this software, it's working great now!

nightsky30, 2019-05-22:
I'm seeing this randomly with my firefly.  Today my firefly shut off the LEDs after about 5 minutes after I set all 3 devices to static blue.  The keyboard and mouse remained lit.  Selecting an effect caused the firefly to light up again.  Now it's been 10 minutes and the firefly is still lit.

There's nothing in the logs indicating an issue.

z3ntu, 2019-05-22:
Sounds like maybe the kernel shuts down the USB port for power saving.

miqued, 2019-05-22:
I've reopened it. I don't think whatever I thought fixed it actually did. I've been away from my pc since I opened this issue and it seems to still be a problem for me. I set everything to a static color as well.

I'll see if it's USB power saving or something else.

nightsky30, 2019-05-22:
@miqued Thanks! :D

@z3ntu I do have my firefly plugged into a bluish USB port on the back of my desktop.  Maybe it gets special power treatment??

nightsky30, 2019-05-23:
So I'm wondering if all ports are set to auto by default, and the reason the firefly port is powered off is because there isn't constant traffic flowing through it when the effect is set to static.  The firefly isn't sending/receiving anything else after the effect is set.  If the firefly effect were set to spectrum or wave, then the traffic should in theory keep the port active and powered.  The keyboard and mouse produce traffic through movement and key press events besides lighting information.  So those ports get enough throughput to remain powered.

Does this sound crazy?  Could a udev rule or some tweak to the daemon set the firefly ../port/power/control setting to "on" when the device is plugged in?

terrycain, 2019-05-23:
So if the firefly is set to wave it shuts off after minute?

nightsky30, 2019-05-23:
So far I've only seen it cut out while on static.

Trying wave now, though I'm not seeing constant traffic on that effect like I thought I would.  Does the device produce the effect on its own?

terrycain, 2019-05-23:
Yeah there are hardware effects which don't do much usb communication. From first glance it looks like the device is idling and being put to sleep

nightsky30, 2019-06-09:
So far if set to spectrum or wave the firefly seems to stay awake.  I've tried static with different colors, and for some strange reason it only seems to sleep when it's in a blue hue.  And that makes no sense.  Maybe there is some other circumstance I'm missing.

@miqued Can you get it to fall asleep with a static color other than a blue hue?

miqued, 2019-06-09:
I've only ever had it on with white

closed 2019-11-06 by stale[bot]