openrazer/openrazer #738: The daemon didn't detect a device that is connected

slipsnip, 2018-12-18:
Interesting, I swore my Orbweaver is plugged in.

kalanihelekunihi, 2018-12-19:
Since the installation of 2.4.0-1~396~ubuntu18.10.1 (latest daily as of 12/19/2018), I can confirm that my Razer devices are no longer detected. I have a Razer Blade 15 Advanced, a BlackWidow Chroma v2 and Mamba Wireless connected to this machine. Checking on my ~/.local/share/openrazer/logs/razer.log, I can see a new message presented on the daemon startup, namely: razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0233.0004/device_type, file is not owned by plugdev

z3ntu, 2018-12-19:
@kalanihelekunihi Did you try rebooting?

@slipsnip Please post the output of the command `sudo dkms install openrazer-driver/2.4.0`

tildearrow, 2018-12-19:
@z3ntu These issues are very common. Why not suggest to reboot after installing the driver?

z3ntu, 2019-03-23:
I'm going to close this issue because there wasn't been a response for over 30 days. Feel free to reply if you are still experiencing this issue.

closed 2019-03-23 by z3ntu