openrazer/openrazer #742: Release on snapcraft

redaxmedia, 2018-12-21:
Sorry to bother you, but I came across [snapcraft.io](https://snapcraft.io) and tough this might be an valid alternative to simplify the installation process? Related to our discussion in #711

> Snaps can be used on all major Linux distributions, including Ubuntu, Linux Mint, Debian and Fedora.

Distribution support:

-  Arch Linux
-  Debian
-  Deepin
-  Elementary OS
-  Fedora
-  GalliumOS
-  KDE Neon
-  Linux Mint
-  Manjaro Linux
-  openSUSE
-  Parrot Security OS
-  Raspbian
-  Solus
-  Ubuntu
-  Zorin OS

z3ntu, 2018-12-21:
From what I could find, it's not possible to package kernel modules via dkms with snaps.

closed 2018-12-21 by z3ntu

redaxmedia, 2019-01-07:
kernel-module-control provides the ability to insert, remove and query kernel modules. This interface gives privileged access to the device.

https://docs.snapcraft.io/the-kernel-module-control-interface/7853

z3ntu, 2019-01-07:
Contributions welcome! :)