openrazer/openrazer #746: Ornata Chroma dead after installing openrazer

jan-r, 2018-12-25:
### Distribution

Linux Mint 18

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hello,
after using my new Ornata Chroma on Win10, I shut down the PC and rebooted into Linux. There I installed the openrazer driver and RazorGenie. It worked right out of the box, and I was able to change the lighting pattern.
After rebooting again, the keyboard was dead. No lights, no key functionality, nothing. The keyboard is now detected as a mass storage device called "CRP1 ENABLD", which contains a single file "firmware.bin". I guess the firmware on the Ornata has crashed and now only runs some kind of emergency boot loader mode to provide some means for a firmware update. But on Windows, the Synapse 2.0 software no longer detects the keyboard and therefore doesn't offer a firmware update.

Any idea what to do now (besides returning the keyboard)? Google turns up nothing but one person having a similar issue (https://www.reddit.com/r/razer/comments/6c810g/blackwidow_chroma_turned_d/) and some references that "CRP1 ENABLD" might be a bootloader mechanism of the NXP controller used in the keyboard.

z3ntu, 2018-12-25:
Out of curiosity, could you send me the firmware.bin file?

But unfortunately for the actual issue I can't help you. Your best bet is probably contacting Razer and not mentioning Linux at all but just describing the issues that show on Windows (extra drive, Synapse doesn't detect it, etc) :wink:

jan-r, 2018-12-25:
The firmware.bin file is a 64 kByte blob of zeros. Hexdump only displays the following:

    0000000 0000 0000 0000 0000 0000 0000 0000 0000
    *
    0010000

I assume that's some kind of reverse-engineering protection of the NXP controller. Usually it shouldn't be possible to read back any flash data. Otherwise it would be too easy to clone the devices.

Yes, I already connected them and avoided mentioning Linux. But I think I'll just return the keyboard via Amazon.

closed 2019-04-02 by z3ntu