openrazer/openrazer #747: Support for Razer Mamba Wireless (1532:0073/0074)

jpg120, 2018-12-26:
### Device Info

* Name: Razer Mamba Wireless
* Product Number: RZ01-02710100
* Model Number: RC30-027101

### Device Information
```
Please post the output of
    lsusb -d 1532: -v 

[lsusbcap.txt](https://github.com/openrazer/openrazer/files/2710489/lsusbcap.txt)

Please post the output of
    ls -lah /dev/input/by-id/

[lscap.txt](https://github.com/openrazer/openrazer/files/2710487/lscap.txt)

### Synapse Screenshots

![calibration](https://user-images.githubusercontent.com/46164980/50450641-c9ab4d00-08f4-11e9-9b5f-5e01614fab05.JPG)
![customize](https://user-images.githubusercontent.com/46164980/50450642-cd3ed400-08f4-11e9-9591-54445e5aab86.JPG)
![lighting](https://user-images.githubusercontent.com/46164980/50450643-cd3ed400-08f4-11e9-9719-ee8cf9af5156.JPG)
![performance](https://user-images.githubusercontent.com/46164980/50450644-cd3ed400-08f4-11e9-9b5b-bffe7f313a67.JPG)
![power](https://user-images.githubusercontent.com/46164980/50450645-cd3ed400-08f4-11e9-820d-e0af35834a06.JPG)

jpg120, 2018-12-26:
That was my fault.  I typo'd in the post and have since corrected.  It is PID 0072.

z3ntu, 2018-12-30:
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we need packet captures.
* And the product & model number of the device for documentation reasons please.

jpg120, 2019-01-01:
I will work on getting the packet captures, but in the mean time, the product and model number follow:
Product: RZ01-02710100
Model: RC30-027101

jpg120, 2019-01-01:
I read through the packet capture guide and think I can get what you need, but I'm not sure exactly what you need to be helpful.  Please let me know what is needed to be useful.

jpg120, 2019-01-02:
Three capture files are attached. 
The DPI change capture files starts at default (1800) then I changed settings to 800, 1800, 4500, 9000, 1800 and then back to 1800.
The color change packet capture file starts at the default (static green).  I changed from that to static red, static blue, breathing red and then concluded with breathing red to blue.
Not sure if the button press file is required, but I performed the following tasks in order:  Left Click, Right Click, mouse 4, Mouse 5, sensitivity down, sensitivity up, scroll click, scroll up, and scroll down.

I hope this helps.  Please let me know if you need anything else.

[Mamba0072captures.zip](https://github.com/openrazer/openrazer/files/2721760/Mamba0072captures.zip)

z3ntu, 2019-01-15:
@jpg120 The pcaps are good, thanks!

@Pajkanon No, this needs implementation now. But as you can probably see there is not much activity from me in this project because 1. I'm working fulltime now (google `Zivildienst` if you're interested^^) and don't have much time/energy to devote to the many projects I'm interested in / working on; and 2. I have a pet project, currently titled [razer_test](https://github.com/z3ntu/razer_test) which I created to solve many fundamental problems with openrazer and am aiming to replace the current openrazer codebase with that, so 99% of the work put into current openrazer will be lost then. :)

gbolcer, 2019-01-31:
Is there a way to just trick it in to thinking it's just a Mamba 2012 Wireless?

bdmayes, 2019-03-21:
@z3ntu Anything we can do to help add this support this either project or razer_test? I'm not exactly sure what to look for in the pcap but if perhaps if someone could point me in the right direction I might be able to help out.

BaronKrause, 2019-05-18:
Same, I have a new Mamba Wireless (Wireless: 91532:0072 Wired: 91532:0073) and if there is any information I can provide or anything I can test to add support for this, please let me know.

nelg, 2019-05-26:
also have this mouse and would really like to reduce the tracking speed.  +1 for this

AzimuthLI, 2019-06-08:
Having the same mouse. It would be amazing to see it work. Very appreaciate for any work here and also be happy to help if anything is requested. +1 for this!

berglh, 2019-07-15:
Same issue, kind of disappointing this has been open for 7 months.
```bash
$ sudo lsusb | grep -i razer
Bus 001 Device 009: ID 1532:0073 Razer USA, Ltd 
Bus 001 Device 004: ID 1532:0072 Razer USA, Ltd
```
I bought this mouse today after seeing the software available on the Razer website, and that this mouse is supposedly supported in that support list, however, not in the control software so far.

@z3ntu Is there anything we can do to add this device?

gbolcer, 2019-07-15:
I had the same issue w/ tracking speed.

xinput list-props 12
xinput list-props 13

(For you it looks like 9 and 4).  

Gave the right details for both the mouse and the USB bluetooth device. 

xset m acceleration threshold

where acceleration defines how many times faster the cursor will move than the default speed. threshold is the velocity required for acceleration to become effective, usually measured in device units per 10ms. acceleration can be a fraction, so if you want to slow down the mouse you can use 1/2, 1/3, 1/4, ... if you want to make it faster you can use 2/1, 3/1, 4/1, ....

I think I also might have turned on natural scrolling.  I put the xset in my .bashrc.  

greg@salt:~/Downloads$ xinput list-props 12
Device 'Razer Razer Mamba Wireless Receiver':
    Device Enabled (153):    1
    Coordinate Transformation Matrix (155):    1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    libinput Natural Scrolling Enabled (287):    0
    libinput Natural Scrolling Enabled Default (288):    0
    libinput Scroll Methods Available (291):    0, 0, 1
    libinput Scroll Method Enabled (292):    0, 0, 0
    libinput Scroll Method Enabled Default (293):    0, 0, 0
    libinput Button Scrolling Button (294):    2
    libinput Button Scrolling Button Default (295):    2
    libinput Middle Emulation Enabled (296):    0
    libinput Middle Emulation Enabled Default (297):    0
    libinput Accel Speed (298):    0.000000
    libinput Accel Speed Default (299):    0.000000
    libinput Accel Profiles Available (300):    1, 1
    libinput Accel Profile Enabled (301):    1, 0
    libinput Accel Profile Enabled Default (302):    1, 0
    libinput Left Handed Enabled (303):    0
    libinput Left Handed Enabled Default (304):    0
    libinput Send Events Modes Available (272):    1, 0
    libinput Send Events Mode Enabled (273):    0, 0
    libinput Send Events Mode Enabled Default (274):    0, 0
    Device Node (275):    "/dev/input/event3"
    Device Product ID (276):    5426, 114
    libinput Drag Lock Buttons (289):    <no items>
    libinput Horizontal Scroll Enabled (290):    1
greg@salt:~/Downloads$ xinput list-props 13
Device 'Razer Razer Mamba Wireless Receiver':
    Device Enabled (153):    1
    Coordinate Transformation Matrix (155):    1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    libinput Natural Scrolling Enabled (287):    0
    libinput Natural Scrolling Enabled Default (288):    0
    libinput Send Events Modes Available (272):    1, 0
    libinput Send Events Mode Enabled (273):    0, 0
    libinput Send Events Mode Enabled Default (274):    0, 0
    Device Node (275):    "/dev/input/event4"
    Device Product ID (276):    5426, 114
    libinput Drag Lock Buttons (289):    <no items>
    libinput Horizontal Scroll Enabled (290):    1

nelg, 2019-07-17:
I discovered that the settings for tracking speed, colours, etc, are stored on the mouse.

So, I plugged it into a windows system with Razers software installed, changed the settings to something I found comfortable, and it kept these settings when plugged back into Linux.

So, a dirty work around, but at least you don't have to throw out the mouse!

Zimm1, 2019-07-20:
> I discovered that the settings for tracking speed, colours, etc, are stored on the mouse.
>
> So, I plugged it into a windows system with Razers software installed, changed the settings to something I [...]

While the settings for dpi, keybindings and macros are stored on the mouse, unfortunately colors and light effects are not.

If you hate the default color profile (like I do), you can disable the light brightness for a profile on Win software, but that's a big bummer. Would love to see my colors working :(

tildearrow, 2019-07-20:
> While the settings for dpi, keybindings and macros are stored on the mouse, unfortunately colors and light effects are not.

I have done effect storage within the daemon in a branch, but I'm still waiting for the Lancehead TE support to be merged in order to push it to upstream...

aldevel, 2019-07-27:
@z3ntu yep, mouse just arrived today and I also had to realize the issue here lmao but not so funny in real. So what is razer_test, is it related to this particular issue? Solution is expected whenever or forget it? Let me know if any information of the system could help something etc. Good luck with your work and other projects anyway!

durzo, 2019-08-28:
I also just purchased this mouse to find its not supported here. shame :(

pedrojimenez, 2019-11-08:
Same problem for me, just purchased the mouse and would like to operate with "openrazer driver" and derivated apps.

`Bus 001 Device 004: ID 1532:0072 Razer USA, Ltd`

pedrojimenez, 2019-11-08:
Any updated on this ?
Could we add any info trying to unlock the Support for the device ?

AbrahamGo, 2019-11-12:
Same issue here, is there any plan to include it?

marin-liovic, 2019-12-02:
what is the status on this issue?

pedrojimenez, 2019-12-02:
I have purchased and just tested a supported "Razer DeathAdder Elite" mouse with Drivers + Polychromatic ... and WooooooW the difference is huge. 

It is difficult to imagine it other way now ...

mbchristoff, 2019-12-26:
Is there progress being made on this topic at this moment?
I'd love to control my mamba wireless together with the rest of my peripherals.

crian, 2020-01-18:
I'm currently working on the implementation of the Mamba Wireless.
So if someone wants to test it here is the link to the source:

[https://github.com/crian/openrazer/tree/feature_mamba_wireless](https://github.com/crian/openrazer/tree/feature_mamba_wireless)

pedrojimenez, 2020-01-20:
I have compiled and tested from the feature "Mamba Wireless" ...
From instructions: https://github.com/openrazer/openrazer/wiki/Building-a-package
Also installed "stable Polychromatic v0.3.12"

Summary:
* Driver loaded on restart (Thaaaaanks !!)
* Battery load wrongly detected (always says 0%)
* Lights effects on logo OK
* Lights effects on scroll not working
* DPI set on zero and does not work
* RATE set on zero and does not work
* Brightness set on zero and does not work

[image]

[image]

crian, 2020-01-20:
@pedrojimenez yes polychromatic isn't working.
I've tested it on razergenie and razercommander and got it working. Maybe you can try one of theses. 
Battery status is still an issue.

durzo, 2020-01-20:
Is anyone interested in starting a prize pool for this?

I would be willing to contribute the cost of a new mamba wireless (in $$ or direct amazon purchase) to anyone who can successfully get this working (with polychromatic) (all features etc)

mbchristoff, 2020-01-20:
@crian Driver loaded using your fork, was not able to control the mouse using polychromatic, razergenie or razercommander though. Every application shows the lighting being in spectrum mode but was not able to change anything.

pedrojimenez, 2020-01-20:
@crian ... confirmed working using RazerGenie too 

Summary:
* Driver also working (great!!)
* DPI manual selection working but not synced when pressing HW buttons
* Effects on Wheel and logo working 
* Polling Rate also working
* Serial Number and Firmware detection not working
* Battery load not found in program (first time used)

[image]

gbolcer, 2020-01-21:
Both old keyboard and mouse work, but no detected devices in RazerGenie or Polychromatic. 

greg@salt:~$ lsusb | grep '1532:'
Bus 005 Device 002: ID 1532:0072 Razer USA, Ltd Razer Mamba Wireless Receiver
Bus 004 Device 002: ID 1532:001a Razer USA, Ltd Razer Marauder

peteisace, 2020-01-29:
@crian Big thanks for this; works very well and blindingly easy to setup (unlike many muddled instructions). Massive props to you and all like you who took the time to help others.

aj53108, 2020-02-02:
@crian I'm a complete noob to Linux.  Just installed my first distro 2 days ago.  I went with Manjaro KDE.  I installed the master openrazer package from the AUR, which obviously is not working with my Mamba Wireless.  I cannot for the life of me figure out how to install your branch. The instructions that were linked a few posts up just say "just build the package from the AUR" but I have no idea how to do that!  Any help from anyone would be greatly appreciated as my mouse is the last thing that I have to get running properly.

crian, 2020-02-03:
@aj53108 just download the [snapshot](https://aur.archlinux.org/cgit/aur.git/snapshot/openrazer-git.tar.gz) of openrazer-git pkg, unpack it, open `PKGBUILD` and edit
`source=("git+https://github.com/openrazer/openrazer.git")`
to
`source("git+https://github.com/crian/openrazer.git#branch=feature_mamba_wireless")`
Then run `makepkg -si`

aj53108, 2020-02-03:
@crian Thank you!!  That did the trick!

marin-liovic, 2020-02-14:
Is that fork getting merged here and released any time soon?

z3ntu, 2020-02-17:
@marin-liovic I don't see a pull request for the Mamba Wireless yet so that means for the time being, no.

nelg, 2020-02-19:
Tested on Mageia 7, using
https://github.com/crian/openrazer/tree/feature_mamba_wireless

Seems to work ok, including polychromatic.

DPI slider works, although moving the DPI up beyond about 1800 makes the
mouse move way to fast.  Maybe just what this mouse does.

Glen

crian, 2020-02-24:
Made a pull request. I'm just waiting for review.
Battery charge status is currently not supported.

pascaldls, 2020-02-28:
@crian how to make it work on mint / ubuntu ?
Sorry for noob question

pascaldls, 2020-03-05:
Support for battery level 
Anyone got this to work

closed 2020-03-18 by z3ntu

CobyBE, 2020-03-30:
Thanks Crian! Send me your paypal and I'll buy you a beer!

marin-liovic, 2020-03-31:
when is this getting released?

z3ntu, 2020-03-31:
@marin-liovic In the next couple of weeks.

crian, 2020-03-31:
Got a fix for battery status. Just waiting for https://github.com/openrazer/openrazer/pull/1055
Thanks to @nightsky30

commit b7f4cef9 (2020-03-18): Add support for Razer Mamba Wireless

Fixes #747