openrazer/openrazer #748: Feature Request: Support Kernel 4.20 LED Pattern Trigger

kalanihelekunihi, 2018-12-26:
### Distribution
Ubuntu 18.04

### Support
Update LED Driver to support Linux Kernel LED pattern triggers.
http://lkml.iu.edu/hypermail/linux/kernel/1810.2/05683.html

With the 4.20 release, the Linux Kernel now includes a standard process for managing colors and patterns of LEDs through the use of an LED Class Driver, Trigger, and Pattern.

z3ntu, 2018-12-27:
Pull requests are welcome :)

But there's also a patch on lkml about a kernel driver for Razer devices that use that interface, I think.
I don't have plans on implementing this myself because I have a fully user space driver in progress right now that will solve many of the current problems hopefully.

kalanihelekunihi, 2018-12-27:
I had not seen the kernel driver under development. Would you mind pointing me in that direction? Also, I’d be very interested in seeing or even helping on the new user space driver as I would like to see devices like the Huntsman Elite supported, as well as some more complex patterns such as wave with ripple on keypress on devices rather than a single pattern.

z3ntu, 2018-12-27:
The subject of themainline kernel driver is "RFC: HID: razer: Add a driver for Razer Keyboards". You should be able to find it on the linux-input mailing list.

The user space driver I'm currently developing is located under https://github.com/z3ntu/razer_test

closed 2019-05-26 by stale[bot]