openrazer/openrazer #749: ModuleNotFoundError: No module named 'six'

ChloeStar18, 2018-12-27:
Using Arch Linux.

 openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-12-27 05:59:15 CET; 41s ago
     Docs: man:openrazer-daemon(8)
  Process: 29991 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 29991 (code=exited, status=1/FAILURE)

Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:   File "/usr/lib/python3.7/site-packages/pyudev/device/__init__.py", line 28, in <module>
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:     from ._device import Attributes
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:   File "/usr/lib/python3.7/site-packages/pyudev/device/_device.py", line 40, in <module>
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:     from pyudev.device._errors import DeviceNotFoundAtPathError
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:   File "/usr/lib/python3.7/site-packages/pyudev/device/_errors.py", line 34, in <module>
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]:     from six import add_metaclass
Dec 27 05:59:15 AllEvil openrazer-daemon[29991]: ModuleNotFoundError: No module named 'six'
Dec 27 05:59:15 AllEvil systemd[640]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Dec 27 05:59:15 AllEvil systemd[640]: openrazer-daemon.service: Failed with result 'exit-code'.
Dec 27 05:59:15 AllEvil systemd[640]: Failed to start Daemon to manage razer devices in userspace.

ChloeStar18, 2018-12-27:
I hope someone can help me on how to resolve this, I couldn't find anything with google.

z3ntu, 2018-12-27:
Filed a bug for the Arch Linux python-pyudev package: https://bugs.archlinux.org/task/61182

As a workaround for now install `python-six` with pacman manually.

closed 2018-12-27 by z3ntu