openrazer/openrazer #751: Support for Razer Goliathus Chroma

multiflexi, 2019-01-03:
### Device Info

* Name: Razer Goliathus Chroma
* Product Number:  RZ02-02500100-R3M1
* Model Number: RZ02-0250

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
[lsusb.txt](https://github.com/openrazer/openrazer/files/2724731/lsusb.txt)

```

```
Please post the output of
    ls -lah /dev/input/by-id/
[lscap.txt](https://github.com/openrazer/openrazer/files/2724733/lscap.txt)

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse> - will add later

Zeroneca, 2019-02-22:
Is there any update on this? Or anyone already working on it? If no, I could try to do this (no guarantee I will succeed)

z3ntu, 2019-03-28:
Support was merged in #796

closed 2019-03-28 by z3ntu