openrazer/openrazer #752: Keyboard lighting Razer blade pro 2017

JasonAsmk, 2019-01-08:
### Distribution
Arch linux 4.19.8-arch1-1-ARCH

### Support
First of all, I'd like to say thank you for your amazing work. I really appreciate all your efforts.
I have a Razer blade pro 2017 which works perfectly and I've been using openrazer through razer-commander and such to change the RGB settings of the keyboard. I've noticed that ripple effects, as well as custom keyboard lighting layouts, don't work. I guess it's not supported yet or probably I'm doing something wrong.
Are there any plans to support this? And if so, can I help somehow with reversing the protocol? I'm a developer with very little experience with drivers and sys programming but I guess I could do that much if needed.

z3ntu, 2019-01-15:
Ripple and custom lighting stuff is mostly software side so there's nothing to reverse engineer :)

What is it about the ripple effect that doesn't work? Is there an error in the logs or does no effect show when choosing the effect in a frontend?

JasonAsmk, 2019-01-16:
So I restarted my pc and tried setting the ripple effect first(with no result, keys not lighting up) and then tried to set some custom layout(in which some keys were correct, others misplaced, others not lighting up). I attached the daemon log and the journalctl output below. It seems that it successfully set the ripple effect although it didn't light up and I think that the custom lighting did not show up at all (probably due to razercommander?). Looks like two separate issues. What's your take on this? I'll certainly lookup razer-commander's issues. Is there a way to invoke openrazer's chromatic effects via scripting or via a command line tool?

This is the Daemon log.
```
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStarlightRandom method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRipple method to DBus
2019-01-16 18:55:37 | razer.device0                  | DEBUG    | Adding razer.device.lighting.custom.setRippleRandomColour method to DBus
2019-01-16 18:55:37 | razer.device0.keymanager       | WARNING  | No event files for KeyWatcher
2019-01-16 18:55:37 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2019-01-16 18:55:37 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2019-01-16 18:55:37 | razer                          | INFO     | Serving DBus
2019-01-16 19:00:50 | razer                          | DEBUG    | DBus called get_serial_list
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_device_type
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_device_name
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_device_type
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_firmware
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call version
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call has_matrix
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call has_matrix
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_keyboard_layout
2019-01-16 19:00:50 | razer.device0                  | DEBUG    | DBus call get_brightness
2019-01-16 19:00:58 | razer.device0                  | DEBUG    | DBus call set_brightness
2019-01-16 19:00:58 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladePro2017FullHD:BY1809A14602282, 'setBrightness', 255)
2019-01-16 19:00:58 | razer.device0                  | DEBUG    | DBus call set_ripple_effect
2019-01-16 19:00:58 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladePro2017FullHD:BY1809A14602282, 'setRipple', dbus.Byte(255), dbus.Byte(255), dbus.Byte(255), dbus.Double(0.05))
2019-01-16 19:01:38 | razer.device0                  | DEBUG    | DBus call set_brightness
2019-01-16 19:01:38 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladePro2017FullHD:BY1809A14602282, 'setBrightness', 255)
```
This is the `sudo journalctl -b | grep -i razer_mount` output
```
Jan 16 18:55:23 midoriHebi razer_mount[455]: Driver razerkbd
[... cut ...]
Jan 16 18:55:23 midoriHebi razer_mount[460]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:022F.0002/
Jan 16 18:55:23 midoriHebi razer_mount[461]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:022F.0001/
Jan 16 18:55:23 midoriHebi razer_mount[465]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022F.0001/
Jan 16 18:55:23 midoriHebi razer_mount[466]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022F.0002/
Jan 16 18:55:23 midoriHebi razer_mount[467]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:022F.0003/
```

z3ntu, 2019-01-23:
> Is there a way to invoke openrazer's chromatic effects via scripting or via a command line tool?

Yes, see https://github.com/openrazer/openrazer/blob/master/examples/custom_zones.py as an example.

z3ntu, 2019-03-23:
I'm going to close this issue because there wasn't been a response for over 30 days. Feel free to reply if you are still experiencing this issue.

closed 2019-03-23 by z3ntu