openrazer/openrazer #753: Daemon to manage razer devices in userspace issues

Sammers21, 2019-01-08:
This happens when i m trying to launch _razergenie_

```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-01-08 22:16:52 MSK; 25s ago
     Docs: man:openrazer-daemon(8)
  Process: 4735 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 4735 (code=exited, status=1/FAILURE)

Jan 08 22:16:52 x1 systemd[2103]: Starting Daemon to manage razer devices in userspace...
Jan 08 22:16:52 x1 openrazer-daemon[4735]: Starting daemon.
Jan 08 22:16:52 x1 openrazer-daemon[4735]: 2019-01-08 22:16:52 | razer                          | CRITICAL | User is not a member of the plugdev group
Jan 08 22:16:52 x1 openrazer-daemon[4735]: Stopping daemon.
Jan 08 22:16:52 x1 systemd[2103]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Jan 08 22:16:52 x1 systemd[2103]: openrazer-daemon.service: Failed with result 'exit-code'.
Jan 08 22:16:52 x1 systemd[2103]: Failed to start Daemon to manage razer devices in userspace.
```

**ENV:**
Laptop: ThinkPad x1 carbon gen6 
OS: Fedora 29 with Linux kernel version: 4.19.13-300
Headsets: Razer Kraken Essential

z3ntu, 2019-01-09:
Please read the error message

> Jan 08 22:16:52 x1 openrazer-daemon[4735]: 2019-01-08 22:16:52 | razer                          | CRITICAL | User is not a member of the plugdev group

and add your user to the plugdev group (see https://github.com/openrazer/openrazer/wiki/Troubleshooting#plugdev)

z3ntu, 2019-03-23:
I'm going to close this issue because there wasn't been a response for over 30 days. Feel free to reply if you are still experiencing this issue.

closed 2019-03-23 by z3ntu