openrazer/openrazer #755: Blackwidow Ultimate 2016 connected, but not detected

LetsLeppr, 2019-01-13:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

LetsLeppr, 2019-01-13:
I am on the latest Manjaro Gnome. Running `ls -l /sys/bus/hid/drivers/razer*/0*/` is giving me 

> ls -l /sys/bus/hid/drivers/razer*/0*/
as an result, but still openrazer doesn't detect it.

z3ntu, 2019-01-15:
That `ls` command can't output the text that you have pasted there, that's the output of the `lsusb` command. Could you please try that again?

StopMotionCuber, 2019-01-20:
I wanted to comment that I also had this issue last week. I don't have this problem anymore, I would guess that it got resolved in an update

z3ntu, 2019-03-23:
I'm going to close this issue because there wasn't been a response for over 30 days. Feel free to reply if you are still experiencing this issue.

closed 2019-03-23 by z3ntu