openrazer/openrazer #760: Unable to lock on pidfile

Masy, 2019-01-23:
### Distribution

Linux Mint 19

### Support
So i have made sure i am a member of the plugdev group
```
 masy@masy-pc-linux ~  $ sudo adduser masy plugdev
[sudo] password for masy: 
The user `masy' is already a member of `plugdev'.
```

and i have rebooted after installing the openrazer deb and still i get this error:
```
openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-01-23 12:42:19 CET; 16s ago
     Docs: man:openrazer-daemon(8)
  Process: 4553 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 4553 (code=exited, status=1/FAILURE)

Jan 23 12:42:19 masy-pc-linux systemd[1578]: Starting Daemon to manage razer devices in userspace...
Jan 23 12:42:19 masy-pc-linux openrazer-daemon[4553]: Unable to lock on the pidfile.
Jan 23 12:42:19 masy-pc-linux systemd[1578]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Jan 23 12:42:19 masy-pc-linux systemd[1578]: openrazer-daemon.service: Failed with result 'exit-code'.
Jan 23 12:42:19 masy-pc-linux systemd[1578]: Failed to start Daemon to manage razer devices in userspace.
```

But the result "exit-code" isn't really something helpful

z3ntu, 2019-01-23:
The `Unable to lock on the pidfile.` message means that the daemon is already running.
Can you please run `pgrep -a -f openrazer` and copy the output here?

Masy, 2019-01-23:
It returns
``3802 openrazer-daemon ``

But the device list is still empty
```
masy@masy-pc-linux ~  $ python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[]
```

z3ntu, 2019-01-23:
Does `systemctl status openrazer-daemon.service` still say "Failed"?

Masy, 2019-01-23:
```
masy@masy-pc-linux ~  $ systemctl status openrazer-daemon.service
Unit openrazer-daemon.service could not be found.
```

z3ntu, 2019-01-23:
Oops I mean `systemctl --user status openrazer-daemon.service`

Masy, 2019-01-23:
Still says failed

Masy, 2019-01-23:
Well, i think i kinda fixed it, but i guess i missed that the Backwidow Elite is not supported yet, so i guess that's why the list of devices is empty