openrazer/openrazer #761: User is not a member of the plugdev group, even though I am.

mjaw3, 2019-01-24:
As the title shows, I am in the plugdev group. However, when I have restarted it still seems to not work.

I have attached the logs and a screenshot.

I have also tried removing all razer deps and polychromatic and reinstalling. I have UEFI turned off in bios, I have tried other razer linux software but still this did not work. Looks like it might be an issue with daemon but I am unsure where to go. Please halp.

[kernel_module.log](https://github.com/openrazer/openrazer/files/2793844/kernel_module.log)
[razer.log](https://github.com/openrazer/openrazer/files/2793845/razer.log)
[razer_mount.log](https://github.com/openrazer/openrazer/files/2793846/razer_mount.log)
[image]

mjaw3, 2019-01-25:
> The end of the log says that the daemon is working properly.
> 
> But the razer_mount log says that the modprobing the driver failed, so it's probably because the kernel headers aren't installed or secure boot is enabled.

I've got secure boot disabled so that should be the case. I'll check the kernel headers, though I'm sure I've downloaded them. Just to double check I'm using "pacaur -S linux414-header" and this still didn't work, is there a different header I need to install?

z3ntu, 2019-01-25:
If `sudo dkms install openrazer-driver/2.4.0` says it's already installed, then run `sudo modprobe razerkbd`. One of those two should error out

mjaw3, 2019-01-25:
> If `sudo dkms install openrazer-driver/2.4.0` says it's already installed, then run `sudo modprobe razerkbd`. One of those two should error out

Ran that command seemed that it couldn't find the directory even tho it was there.

Checked in Octopi and just installed all the linux lts headers inside there.

[image]

Seems that just installing all the linux lts headers from there fixed the issue.

Thanks for the help :) It's all running good now :dancer:

z3ntu, 2019-01-25:
Great!

closed 2019-01-25 by z3ntu