openrazer/openrazer #763: Can not run daemon on Manjaro Linux 18.0.2

JHSeng, 2019-01-28:
I have tried rebooting computer after installing openrazer and polychromatic. But it's still not working.

[image]
[image]

what's wrong with it?

lah7, 2019-01-28:
Your user account needs to be added to the `plugdev` group, like so:

    sudo useradd -G plugdev $(whoami)

Then log out and log back in to take effect.

JHSeng, 2019-01-28:
> Your user account needs to be added to the `plugdev` group, like so:
> 
> ```
> sudo useradd -G plugdev $(whoami)
> ```
> Then log out and log back in to take effect.

thanks, the problem is solved.

closed 2019-01-28 by z3ntu