openrazer/openrazer #769: daemon not found // couldn't connect to daemon device manager!

lowao, 2019-02-05:
Not too sure how to correct this.
output of polychromatic-controller https://pastebin.com/nbSK7gXn
Troubleshooting: https://pastebin.com/yiXk8v5G

lah7, 2019-02-05:
A number of those issues were obsolete, since `razer-daemon` and `razer-service` are from a very old codebase. The executable is `openrazer-daemon`, so:

    openrazer-daemon -Fv

Would run it in the foreground and verbose.

From what I can see, your installation of the OpenRazer's Python library has a syntax error:
```
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/daemon.py", line 28, in <module>
    import openrazer_daemon.hardware
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/hardware/__init__.py", line 5, in <module>
    from openrazer_daemon.hardware.device_base import RazerDevice
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/hardware/device_base.py", line 483
    diff --git a/daemon/openrazer_daemon/hardware/device_base.py b/daemon/openrazer_daemon/hardware/dev
               ^
SyntaxError: invalid syntax
```

Somehow, you've got a command appended at the end of the file (line 483)
> `diff --git a/daemon/openrazer_daemon/hardware/device_base.py `

Please remove it from `/usr/lib/python3.7/site-packages/openrazer_daemon/hardware/device_base.py` and see if it works. Or completely reinstall the OpenRazer python library.

lowao, 2019-02-06:
Removing line 483 didn't help.
https://pastebin.com/FPQMU3p3

lah7, 2019-02-06:
The problem was precisely due to some output (looks like from `diff`) was echoed/appended at the end.

Which device are you using? OpenRazer only _officially_ supports M1-M5 macro keys on keyboards right now. There's still work needed to properly support key mapping.

For profiles, that would be something a front-end would do. Polychromatic has some new effects/profile features in development for a future update.

lowao, 2019-02-07:
The previously issue was probably linked to the diff line, right? Was that added by openrazer drivers or did I manually add that? I don't think I did, but I could never get the daemon started before at any point.

Ooops, I forgot to provide some very important information. I am using the Orbweaver Chroma. On the github, it looks like Orbweaver is supported. Are there limitations to the orbweaver as well?

lah7, 2019-02-07:
> Was that added by openrazer drivers or did I manually add that?

It was definitely a user error, but accidents happen!

> Are there limitations to the orbweaver as well?

You can change lighting on the **Orbweaver Chroma**, but there's no key remapping unfortunately. See https://github.com/openrazer/openrazer/issues/744#issuecomment-449580502 and https://github.com/openrazer/openrazer/issues/614#issuecomment-429407337. There are other tools that others got to work like [Keyboard Master](https://bitbucket.org/conzar/keyboarding-master/src).

I have a Orbweaver Chroma too, but out of the box, it only behaves like the top-left portion of the keyboard once the daemon initialises it.

lowao, 2019-02-07:
Oops. I fat fingered it. I guess. Thanks for the support though. Is there anyway to change the profile on Windows and bring that over? Or change the default profile, and maybe that would carry over?

I'll look into Keyboard Master in the mean time. Thanks for the suggestion.  I'm kind of surprised I didn't find the lack of key mapping support during my research. I probably just assumed that since it was mentioned on the main openrazer github's page that all was fine and dandy.

Thanks again for the help.

lah7, 2019-02-07:
> Is there anyway to change the profile on Windows and bring that over? Or change the default profile, and maybe that would carry over?

For lighting, possibly (it depends on firmware) but a definite no for remapped keys. Since key maps are not stored on hardware, it's the software / driver's job (Synapse) to translate key presses to specific keys and playback macros.

There is [a project](https://github.com/z3ntu/razer_test) from one of the maintainers to bring OpenRazer into userspace (so no driver, and in theory will be easier to handle special functionality like remapping keys). For now, lighting is our strength.

We should probably update the wiki / website to be clear what is/isn't supported.

TherinS, 2019-02-09:
Its great to see people making projects for the Tartarus/Orbweaver and Belkin gameboards, and Conzar's KeyboardingMaster works nice. However, if you want to get input working and macro support NOW, I would recommend looking at Pystromo. It has lots of configuration available including macro support. It has been around since 2008 but has no GUI. The configuration files are human readable, however, and it has worked great for me for years, until I made the fateful decision to upgrade from Ubuntu 16.04 to 18.04.  I have not tested it with a non-Ubuntu based distro.

The v0.6.0 no longer remaps my keys, though it DOES seem to work, as evidenced by the 3 LED keys changing state. The v0.6.1 had the lib files upgraded to work with Python3 (or so the dev claims) but I cannot get the remapper to run, as the terminal returns several 'unable to load module' errors.

I think the OpenRazer project could get instant Nostromo/Tartarus/Orbweaver key re-binding and macro support using Pystromo and would only need to make a GUI to populate the .map files, if only the program worked in 18.04. Since it is entirely Python scripts and just about all Linux distros have Python installed, the actual backend files are less than 100kb (!), not including configuration map files.

If anyone knows Python and what changed between 16.04 and 18.04, please let me know what I need to do to get this wonderful bunch of code working again, even without any GUI. I have made install and activation scripts to streamline using Pystromo and it starts up nearly instantly.

v0.6.0
https://github.com/byrongibson/Pystromo

v0.6.1 (Python3 compliant? Fails to run for me)
https://github.com/booms8/Pystromo

lowao, 2019-02-12:
I went over to xkbcomp, change keymapping for any keyboard with individual support. It's pretty easy to setup with a bit of googling. I do however have once issue with it, but the program doesn't work with a java application, but I am still troubleshooting it. Other than that, works great. I would recommend checking it out, TherinS.

Custom Profile Already Configured for the OrbWeaver with the use of xkbcomp
https://github.com/GuillaumeSeren/orbweaver-xkb

z3ntu, 2019-02-24:
Closing this as the issue seems to be solved.

closed 2019-02-24 by z3ntu

chabad360, 2020-04-19:
@ecayoume I've been dropping this around recently, [tildearrow/input-modifer](https://github.com/tildearrow/input-modifier) might be another good option to look at.

z3ntu, 2020-04-20:
@chabad360 Please don't necro-bump issues.