openrazer/openrazer #771: Repository key expired at Debian 9.0

enyone, 2019-02-08:
```
$ curl -s http://download.opensuse.org/repositories/hardware:/razer/Debian_9.0/Release.key | gpg --with-fingerprint
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048 2014-09-16 [SC] [expired: 2019-02-01]
uid           hardware OBS Project <hardware@build.opensuse.org>
```

batmanm0b1E, 2019-02-08:
oK somebody made this thanks for opening this, I'm having the same problem too

z3ntu, 2019-02-08:
Could you please try again now?

kenzierocks, 2019-02-08:
Looks good on my end:
```
$ curl -s http://download.opensuse.org/repositories/hardware:/razer/Debian_9.0/Release.key \
 | gpg --with-fingerprint --show-key
pub   rsa2048 2014-09-16 [SC] [expires: 2021-03-28]
      E96C 496E 4E77 C159 C2FC  A053 062F 9FD4 D6D1 1CE4
uid                      hardware OBS Project <hardware@build.opensuse.org>
```

However, xUbuntu 18.04, 18.10 still needs the new key:
```
curl -s http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_18.10/Release.key \
 | gpg --with-fingerprint --show-key
pub   rsa2048 2014-09-16 [SC] [expired: 2019-02-01]
      E96C 496E 4E77 C159 C2FC  A053 062F 9FD4 D6D1 1CE4
uid                      hardware OBS Project <hardware@build.opensuse.org>
```
(same result for 18.04)

batmanm0b1E, 2019-02-09:
Its working for me Thanks. Using Debian 9.0 repo. You should be able to fix Ubuntu key easily I'm not on it so i cannot check sorry

enyone, 2019-02-09:
```
$ curl -s http://download.opensuse.org/repositories/hardware:/razer/Debian_9.0/Release.key \
>  | gpg --with-fingerprint
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048 2014-09-16 [SC] [expires: 2021-03-28]
uid           hardware OBS Project <hardware@build.opensuse.org>
```

z3ntu, 2019-02-09:
I'd say it's fixed now.

closed 2019-02-09 by z3ntu