openrazer/openrazer #774: Support for [device name here]

pkaramol, 2019-02-11:
### Device Info

* Name: RAZER Blackwidow ELITE
* Product Number: RZ03-02621500-R3P1
* Model Number: RZ03-0262

### Device Information
```
$ lsusb | grep '1532:'
Bus 001 Device 008: ID 1532:0228 Razer USA, Ltd 
```

```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 140 Feb 11 10:05 .
drwxr-xr-x 4 root root 560 Feb 11 10:05 ..
lrwxrwxrwx 1 root root   9 Feb  8 13:18 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Feb  8 13:18 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 Feb 11 10:05 usb-Razer_Razer_BlackWidow_Elite-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Feb 11 10:05 usb-Razer_Razer_BlackWidow_Elite-if01-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 Feb  8 13:18 usb-SunplusIT_Inc_HP_HD_Camera-event-if00 -> ../event17

```

BTW is there a way to disable / enable  keycaps lightning on Ubunytu 18.04?

antonis179, 2019-02-14:
Duplicate of #684

pkaramol, 2019-02-14:
OK closing then this one as it is a duplicate;

closed 2019-02-14 by pkaramol