openrazer/openrazer #775: Support for Razer Blade 15” (2019) - Advanced

rsimmonsjr, 2019-02-11:
### Device Info

* Name: Razer Blade 15” (2019) - Advanced
* Product Number: RZ09-2888E92
* Model Number: No information

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 003: ID 1532:023a Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x023a 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 180 Feb 10 22:44 .
drwxr-xr-x 4 root root 600 Feb 10 22:44 ..
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Azurewave_Integrated_Camera_200901010001-event-if00 -> ../event3
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Azurewave_Integrated_Camera_200901010001-event-if02 -> ../event4
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Razer_Razer_Blade-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Razer_Razer_Blade-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Razer_Razer_Blade-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root  10 Feb 10 22:44 usb-Razer_Razer_Blade-if02-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 Feb 10 22:44 usb-Razer_Razer_Blade-if02-mouse -> ../mouse2
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

closed 2019-03-19 by z3ntu

arty-hlr, 2019-04-09:
I see this has been closed, but was support the blade stealth 2019 added? Mine still doesn't get recognized.

Thanks for the great work, it's great to have such a community behind porting razer to linux :)

mgruppi, 2019-04-10:
Hello. Support for the Blade Stealth 2019 is available. Is it the standard version of the Stealth or some special editions such as Mercury or Quartz? You may wanna check it by running `lsusb | grep '1532:'` and checking if the output matches the VID:PID in the main page of this github repo. I'm assuming these special versions might have a different PID, but it may not be the case.

Also, you may wanna try troubleshooting: https://github.com/openrazer/openrazer/wiki/Troubleshooting

arty-hlr, 2019-04-10:
it is the standard version, and indeed the VID:PID matches one in the main page of the repo, so the problem is not that. I'll go into troubleshooting, thanks!

commit 1be60bba (2019-03-19): Razer Blade 2019 advanced support (#773)

Fixes #775