openrazer/openrazer #776: Issue with the SteamGame Razor2: Hidden Skies (SteamAPPID:34920) and Razer Blackwidow Chroma V2

UserohneNamen, 2019-02-14:
### Distribution

Linux Mint 19.1 Cinnamon (Kernel 4.15.0-45-generic)

### Support

- [Y] Have you tried rebooting?
- [Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I dont know why but every time i Start this game ( https://store.steampowered.com/app/34920/Razor2_Hidden_Skies/ ) and reach the Main-Menu of the Game, my Keyboard (Razer BlackWidow Chroma V2) shuts down and wont respond until i restart my PC. I have also described this issue on the ProtonDB. This is the first and only Game this happend. Dont know why.

Sorry for my bad english. Im german. Have a great day! :]

z3ntu, 2019-02-14:
Does the issue exist also when OpenRazer is not installed? If that's the case, it's not our problem but for other peoples' :wink:

UserohneNamen, 2019-02-14:
I will try and let you know ;] And Thanks for the fast reply! :D

UserohneNamen, 2019-02-14:
OK, i tryed it with the same result. It also shuts down if openrazer isnt installed. Still very suspicious on the Games-side. I never had such an impact on my Hardware with ANY other game before... However - like you say, its not the problem of OpenRazer so... This issue can be closed. Sorry.

z3ntu, 2019-02-14:
Good luck with finding a fix :)

closed 2019-02-14 by z3ntu