openrazer/openrazer #777: Deathstalker Chroma coil whine when lit up

rhinus, 2019-02-16:
I am running the most current version of ubuntu, as well as the most current version of openrazer.
On windows the keyboard is fine, it emits no sound and everything works.
On ubuntu though, everything works great except that when the keyboard is lit up it emits high frequency sound like coil whine.

z3ntu, 2019-02-16:
Does the issue happen when OpenRazer is not installed?

rhinus, 2019-02-16:
Uninstalling doesn't seem to have a noticeable difference.
Adjusting the brightness adjusts the volume of the whine.
Also there are some colors that the whine will go away for.

closed 2019-02-16 by rhinus

lah7, 2019-02-17:
I too can tune my ears to high pitch sound exactly as you've described. It's definitely the hardware. My keyboard, mouse and mousemat do it even on an entirely different system - Synapse for Mac.