openrazer/openrazer #782: Why is the DeathAdder 2013 Defined in "mouse.py" but not finding the mouse?

AnzoTheGamer, 2019-02-21:
The mouse is in the script and fake_driver area - Yet its not picking up on the program
Its even detected in the system:

Outout 

lsusb:
`$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0480:a202 Toshiba America Inc Canvio Basics HDD
Bus 002 Device 002: ID 0480:a006 Toshiba America Inc External Disk 1.5TB
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:0a1f Logitech, Inc. G930
Bus 001 Device 003: ID 0b05:1825 ASUSTek Computer, Inc. 
Bus 001 Device 009: ID 1532:0037 Razer USA, Ltd 
Bus 001 Device 006: ID 1532:0216 Razer USA, Ltd 
Bus 001 Device 004: ID 056a:033e Wacom Co., Ltd 
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 007: ID 0c45:6341 Microdia Defender G-Lens 2577 HD720p Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

z3ntu, 2019-02-24:
Did you look through the [Troubleshooting wiki page](https://github.com/openrazer/openrazer/wiki/Troubleshooting)?

Also what exactly are you trying to achieve?

z3ntu, 2019-03-23:
I'm going to close this issue because there wasn't been a response for over 20 days. Feel free to reply if you are still experiencing this issue.

closed 2019-03-23 by z3ntu