openrazer/openrazer #783: Support for LIFTOFF distance feature in Razer Mouses

leimh, 2019-02-24:
### Device Info

* Name: Razer DeathAdder Elite & others.

### Synapse Screenshots
[image]

It should be interesting to implement this feature for help people who got tracking problems with the default settings.

PS: Currently using synapse, this feature only really works when a Razer MouseMat is selected (not when you set a custom one, this could be a synapse bug): a way to solve this is when you have selected a Razer MouseMAT and have changed the liftoff, then you can select a custom mousemat profile and enjoy the new liftoff setting with it.

Regards.

z3ntu, 2019-02-24:
Could you please record a packet capture of the USB traffic when you adjust the slider in Synapse? See https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol how

leimh, 2019-02-25:
Well, I did it with a low profile laptop, hopefully it works as it should...
I did the captures doing the following sequence:

From 1to100 (i do the same test 2 times like is suggested in the wiki) I only change from one mousemat profile to another without changing the liftoff distance. ( I don't know if the diference from a surface to another is something "cosmetic" or is translated directly in a diference in the led power level that is lightning that surface or/and are more parameters involved)

From 100to131 i change the liftoff parameter using one mousemat profile and between very different profiles (hard mousemat vs textile mousemat).
I added all the mousemats following the order in which their appears in razer synapse menu with the exception of vespulas that are the lastests ones (between the razer mousemats). The option to add a custom mousemat profile is always the lastest one. 

Pcapng captures & Synapse screenshots:
https://uploadfiles.io/y9kqo

Regards.

z3ntu, 2019-02-25:
I actually see packets that are look pretty similar but are a bit different in each capture. I'll see what I can do with them.

leimh, 2019-02-26:
I redid some test trying to be faster (to reduce "the size of the captures"), and i added a first capture trial changing the colour of the "ambient light" of the mouse that is a supported feature, to validate the capture method.

https://ufile.io/j23e1

Regards.

leimh, 2019-03-07:
Well, i add here more info that could be interesting:
General Datasheet of PMW3389DM-T3QU (Deathadder Elite's sensor with custom razer firmware) can be downloaded from:
http://www.pixart.com/products-detail/4/PMW3389DM-T3QU

General Datasheet of PMW3360 (similar to the previous one with few differences):
http://www.pixart.com/products-detail/10/PMW3360DM-T2QU
More detailed datasheet of PMW3360:
https://d3s5r33r268y59.cloudfront.net/datasheets/9604/2017-05-07-18-19-11/PMS0058-PMW3360DM-T2QU-DS-R1.50-26092016._20161202173741.pdf

z3ntu, 2019-03-20:
@leimh: Here are the values that are parsed out of the pcap: https://pastebin.com/raw/46WqHPme (relevant stuff is in the `Args`)

Have fun figuring out what is what ;)

leimh, 2019-03-24:
@z3ntu: I'm on it, I have my little little hypothesis and I need a little bit of time to check/discard some things (Mainly in the custom calibration part).

leimh, 2019-03-30:
Ok, I'll try to summarize all:

1) I tried to see if any other settings like dpi/mouse_hz changes the liftoff/calibration values and as result they seems to be independent.

2) The "precalibrated" razer settings for their mousemats really are the default surface calibration settings but with the liftoff levels normalized. For example, lets say that the level5 in "X" Razer mousemat correspond to 12mm of liftoff distance, then we can say that on ALL Razer mousemats the level at 5 means 12mm of liftoff (razer called it like "liftoff offset").

3) The Custom profiles works exactly the same like the "precalibrated" part for all levels except the first one. For levels (2-10) it works like a sensor liftoff offset with default surface calibration values. In the case of level at 1, razer synapse uses the calibrated surface values (that the mouse send to it) and force the liftoff offset to a low value (0x10) that is always the same. Lets say that this improve the mouse tracking for a determined surface but only when you set it at level 1. For the other levels (2-10) the mouse send the liftoff offset for the level 1, and with it and maybe other parameters (TODO), synapse calculated the liftoff offset for levels [2-10].

4) What are the surface calibration and liftoff offset values? 
Datasheet Register name (name in Razer Synapse config files) (default value)(max value in hex)
a) Rawdata_Threshold (Param1) (0x0A) (0x3F) 
Really the max value here is the recomended setting that the mouse send after a sucessful calibration process ( LiftCutoff_Cal2 or Liftcutoff_Tune2 ---> RPTH[6:0] ) to rewrite the rawdata_threshold register. 
b) Min_SQ_Run (Param2) (0x10) (0xFF)
c) "Liftoff offset". (Param7) 
Here i don't now which register or combination of registers is this one after seeing the datasheets. In the 3389 datasheet exists a Lift_Config that sets the mouse at 2mm or 3mm. The liftoff ofset could be in relation at this parameter... or/and could be a "new" "shutter" parameter or directly changes the current of the ilumination led,etc.. Anyway it dont really care, all suggest in the direction that lower values makes the liftoff higher. (In the old mice withouth calibration, usually higher ilumination (inside a limits) means higher liftoff distance, then i like the aproach to see the Param7 as the current that drives the ilumination led, besides it really dont adjust to the reality. Anyway searching in internet i find the values or parameters that logitech uses for their similar mice:

```
"surfaceTuning" : 
  {
    "Logitech.Gaming.Mouse.G303.Daedalus.Apex" : 
    {
      "{8d56eb33-30d4-421b-8058-17ca8f582f31}" : 
      {
        "ledDrive" : 0,
        "minSqual" : 255,
        "name" : "Puretrak",
        "pixelThreshold" : 5,
        "time" : 1456507907
      }
    }
  },
```
The razer equivalent archives are for the precalibrated values. Full table at :
"C:\ProgramData/Razer/Synapse3/Service/Staging/Devices/DeathAdder Elite/Bin/mmdatav3.ini"
```

[Razer Invicta Speed 1]
Param0=0x0A
Param1=0x10
Param2=0x83
Param3=0x00
Param4=0x00
Param5=00
Param6=0
Param7=64
Param8=0
[Razer Invicta Speed 2]
Param0=0x0A
Param1=0x10
Param2=0x83
Param3=0x00
Param4=0x00
Param5=00
Param6=0
Param7=59
Param8=0
...
```
For the custom profiles (values in decimal notation)(values send by the mouse after calibration):
"C:\ProgramData/Razer/Razer Central/Accounts/RZR_number_razer_id_account/Emily3/Devices/number_id_device/Calibrations/PixartCalibrationData.xml"

`<PixartCalibration><CustomCalibrationData><PixartCalibrationData><Id>id_mousematprofile1</Id><SensorData><Param0>24</Param0><Param1>234</Param1><Param7>82</Param7></SensorData></PixartCalibrationData><PixartCalibrationData><Id>id_mousematprofile2</Id><SensorData><Param0>31</Param0><Param1>174</Param1><Param7>84</Param7></SensorData></PixartCalibrationData></CustomCalibrationData></PixartCalibration>`

5) Exist 2 calibrations profiles archives based in the maker of the mouse sensor: philips or pixart.

6) If anyone want to check the pcap file and follow the process, i used the following wireshark filter to summarize all the process (i have not experience using it... probably could be improved) :
```
(usb.bus_id == 1 && usb.device_address ==4 && (usb.data_fragment contains 00:04 || usb.control.Response contains 00:04)) || (((usb.transfer_type == 0x01) && (frame.len >= 69)) && ((usb.capdata >= 01:00:00:00:00:00:00:00
) || ((usb.capdata >= 01:00:00:00:00
) && (usb.capdata < 02:00:00:00:00

)) || (usb.capdata == 00:00:00:00:00

) || (usb.capdata == 00:00:00:00:00:00:00:00
)))
```
and other one more simple to capture the packets on the fly and make the following table. The precalibrated values should be the same as the ones that appear in mmdatav3.ini

leimh, 2019-03-30:
@z3ntu I'll explain briefly how custom calib works (or at least how I think it works :P ) using the previous pcaps because if you was able to isolate the calibration data probably you dont need more explanations, but anyway if you need more info just ask me. 
Custom calibration (more info of the process itself from user point of view in: https://www.youtube.com/watch?v=ncNzaktQYFY )(In youtube you can find the surface calibration process of other mice like the Logitech ones, corsair, etc. that could be interesting to see).

**Status Trans Packet Proto DataSize Class CMD Args : 90_create_second_custom_lvl_default_3.pcapng**
[image]
IRQ Left mouse button at 01*
IRQ Left mouse button at 00*

[image]
00     1f    0000   00    03       0b    03  000401 |  Enable mouse calibration operations
IRQ Left Mouse button at 01*
00     1f    0000   00    04       0b    09  00040000 | Start the data acquisition
IRQ at 050A00* | 050A00 means "surface" data acquisition running. 050A0Z, z=0 (running),z=1 (success), z=2 (failed), z=3 (reset mouse calibration after pressing 3 buttons) ~~ RUN_CAL
IRQ Left Mouse Button at 00*

[image]
IRQ at 050A01*  |   successful "surface" data acquisition.
IRQ Left Mouse Button at 01*
00     1f    0000   00    0b       0b    85  0004000000000000000000 | Ask mouse for calibration data
02     1f    0000   00    0b       0b    85  000418f002000000000051 | Mouse sends synapse the params
param1=18,param2=f0,param7=51. The "param3" is always 0x83 in settings using the default param1 and param2; and "param3" is always 0x02 in settings using the surface calibration data sends by the mouse after the calibration process.
00     1f    0000   00    03       0b    03  000401 | "Prepares mouse for mouse calib operation ? or finish previous mouse operations?¿?"
IRQ Left Mouse button at 00*

[image]
00     1f    0000   00    0b       0b    05  00040a1083000000004000 | Sets params values that shows the gui. If was set to level1, param1 and param2 are set to the values that the mouse sends.
00     1f    0000   00    03       0b    03  000401 |  I dont know if this is really needed (making a comparison when the mouse is set using the precalibrated razer profiles)

ScreenShot of the gui when data adq goes wrong:
[image]

TODO (upload reset 3 buttons pcap, calib error pcap, and maybe some to others to check time/waits between some synapse to mouse microcontroller commands).

Regards.

leimh, 2019-04-11:
All_type_of_pcaps (fail at custom calibration & reset pressing 3 buttons included):
https://ufile.io/czodx
(The reset pressing 3 buttons consist in delete all custom profiles that razer stores in ""C:\ProgramData/Razer/Razer Central/Accounts/RZR_number_razer_id_account/Emily3/Devices/number_id_device/Calibrations/PixartCalibrationData.xml" ( and in cloud storage ?¿? ) and send to the mouse the same command that synapse uses for set it with the default calibration settings).

About the previous TODO (know the exact function that synapse uses), just say that is not something that we really need for implement in an exact way because are just different lift off levels (not directly related with ""surface quality"" settings (param0 & param1)). A proportional division between the max recommend param7 value and the minimum value observed in the tables (0x03) could be a good solution. Anyway i think that is more a problem for the frondtend app and how exactly you want to replicate the synapse behaviour but is not needed in terms of funcionality as i'm saying. The only problem could be setting the mouse with a liftoff level that makes that the mouse dont track at all, but this usually happens with the lowest liftoff distance (highest param7 value). Then we are safe using the aprox. and on the other hand, at least the DeathAdder Elite behaviour is reset calibration to defaults when you disconnect/reconnect the mouse.
 
And yep, help is needed 'cause I tried to implement it myself but i guess that my coding skill is 2 low, and it will take me ages if i have to do it :/ 
```
#include "razermousecalibration.h"

/* For DAE: Pixart 3389 sensor and NXP LPC11U35F microcontroller */
/* For Philips sensors we probably need pcaps ?¿?¿? */ 

/*
* Precalibrated razer mousemat settings (param7 ~~ liftoff offset values)
* (Probably better solution using razer files with the frontend app than hardcode here all precalib tables)
*
* static unsigned char destructor_2_mm[] = { level1, level2, ...., level10 }
*
*
* static unsigned char destructor_2_mm[] = { 0x3A, 0x38, 0x34, 0x28, 0x24, 0x1F, 0x1A, 0x16, 0x14, 0x12 }
* static unsigned char gigantus_mm[]
* static unsigned char sphex_v2_mm[]
* static unsigned char custom_qck_mm[]
*/

/* ?¿?¿?¿?¿?¿?¿?¿?¿??¿?¿?¿??¿?¿?¿?
*
* Status Trans Packet Proto DataSize Class CMD Args
* 00     1f    0000   00    0b       0f    03  00 00 00 00 01 80 00 80 80 00 80 | ?¿?¿??¿? (without control response data ?¿?¿?)
*
* 00     1f    0000   00    06       0f    02  00 00 08 00 00 00 | ?¿??¿??¿
*/

/* Status Trans Packet Proto DataSize Class CMD Args
*  00     1f    0000   00    03       0b    03  000401                     | SET CALIB MODE (MODE ON) ~~ ENABLE CALIB OPERATION
*
*  00     1f    0000   00    03       0b    03  000400                     | SET CALIB MODE (MODE OFF) ~~ SET CALIB TO DEFAULTS
*/

struct razer_report razer_calib_set_mode(unsigned char mode)
{
        struct razer_report report = get_razer_report(0x0B, 0x03, 0x03);
        report.arguments[0] = 0x00;
        report.arguments[1] = 0x04;
        report.arguments[2] = mode;

        return report;
}

/*Status Trans Packet Proto DataSize Class CMD Args
* 00     1f    0000   00    0b       0b    05  00040A1083000000002800 | set mouse with calibration data
*
* param0 = 0A, param1 = 10, param2 = 83, param7= 28
*
* param0 = argument[2], param1 = argument[3], param2 = argument[4], param7 = argument[9]
*/

struct razer_report razer_calib_set_parameters(unsigned char param0, unsigned char param1, unsigned char param2, unsigned char param7)
{
        struct razer_report report = get_razer_report(0x0B, 0x05, 0x0B);
        report.arguments[0] = 0x00;
        report.arguments[1] = 0x04;
        report.arguments[2] = param0;
        report.arguments[3] = param1;
        report.arguments[4] = param2;
        report.arguments[5] = 0x00;
        report.arguments[6] = 0x00;
        report.arguments[7] = 0x00;
        report.arguments[8] = 0x00;
        report.arguments[9] = param7;
        report.arguments[10] = 0x00;

        return report;
}

/*Status Trans Packet Proto DataSize Class CMD Args
* 00     1f    0000   00    04       0b    09  00040000 | start surface data acq.
*/

/*********
struct razer_report razer_calib_start_surface_data_acquisition(void)
{
        struct razer_report report = get_razer_report(0x0B, 0x09, 0x04);
        report.arguments[0] = 0x00;
        report.arguments[1] = 0x04;
        report.arguments[2] = 0x00;
        report.arguments[3] = 0x00;

        return report;
}
*********/

/*Status Trans Packet Proto DataSize Class CMD Args
* 00     1f    0000   00    0b       0b    85  0004000000000000000000 | ask for surface data
*
* 02     1f    0000   00    0b       0b    85  000418EA02000000000058 | mouse response with surface data (recommended values).
*
* param0 = 18, param1 = EA, param2 = 02, param7= 58
*
* param0 = argument[2], param1 = argument[3], param2 = argument[4], param7 = argument[10]
*/

/**********
struct razer_report razer_calib_get_surface_data_acquisition(void)
{
        struct razer_report report = get_razer_report(0x0B, 0x85, 0x0B);
        report.arguments[0] = 0x00;
        report.arguments[1] = 0x04;
        report.arguments[2] = 0x00;
        report.arguments[3] = 0x00;
        report.arguments[4] = 0x00;
        report.arguments[5] = 0x00;
        report.arguments[6] = 0x00;
        report.arguments[7] = 0x00;
        report.arguments[8] = 0x00;
        report.arguments[9] = 0x00;
        report.arguments[10] = 0x00;

        struct razer_report response = razer_send_payload(usb_dev, &report);

        return sprintf(buf, "%d:%d:%d:%d\n", response.arguments[2], response.arguments[3], response.arguments[4], response.arguments[10]);
}
**********/
```

Regards.

leimh, 2019-04-11:
@z3ntu In https://github.com/openrazer/openrazer/wiki/Unknown-commands

0x0b | 0x03 | args either 000400 or 000401 or 000500 or 000501 - calibration stuff
-- | -- | --

What device (or synapse version) has generated those 000500 and 000501 arguments?

and on the other hand, do you know what the following commands do?
```
* Status Trans Packet Proto DataSize Class CMD Args
* 00     1f    0000   00    0b       0f    03  00 00 00 00 01 80 00 80 80 00 80 | ?¿?¿??¿? (without control response data ?¿?¿?)
* 00     1f    0000   00    06       0f    02  00 00 08 00 00 00 | ?¿?

```
Regards.

z3ntu, 2019-04-11:
I have some `000500` in Naga 2014; and `000501` and `000500` in Naga Hex V2 packet captures

z3ntu, 2019-04-11:
Also regarding the pastebin from above: I've updated it here (https://pastebin.com/raw/L5rL2Lx4) for the 0b 85 requests with the values that are returned from the device - the rest should be the same.

And I've uploaded the `RazerCalibration.ini` file used by Synapse (2) for my DeathAdder Chroma: [RazerCalibration.ini](https://github.com/openrazer/openrazer/files/3069172/RazerCalibration.ini.txt), maybe it can help you :)

leimh, 2019-04-11:
Checking the following links & "googling" for more info:
https://www.overclock.net/forum/375-mice/854100-gaming-mouse-sensor-list.html
https://www.overclock.net/forum/375-mice/1533712-gaming-mouse-mcu-list.html

We got that:
Razer Naga 2014 (8200dpi): Philips Twin Eye PLN2034 & NXP LPC11U23F (Arm M0) microcontroller
Razer Naga Hex v2 (16000dpi): Philips Twin Eye PLN2037 ?¿?

0x0b | 0x03 | args either 000500 or 000501 - could be ""enable/disable"" calibration in a Philips sensor
-- | -- | --

To know more will be interesting to have pcaps modifying the calib settings & the config synapse files for any of those mice with a philips sensor:

Razer |  
-- | --
Imperator | PLN2032
Imperator 4G | PLN2033
Lachesis | PLN2031
Lachesis 3.5G | PLN2032
Mamba | PLN2032
Mamba 4G | PLN2033
Naga/Hex/Epic/Molten/2012 | PLN2032
Naga 2013/2014 | PLN2034 |  
Orochi | PLN2031
Orochi 2013 | PLN2033
Spectre | PLN2032 |  
Tron | PLN2032 |  

(And 5g laser sensor 16000dpi PLN2037: naga hex v2, mamba te?¿?, etc)
(The ones supported by openrazer)
For this type of mice (sensor) i think the calibration (with own surface data adquisition) is an important feature for all the tracking problems that people seems to have with them. It could help to mitigate them.
(In the case of 5g Infrared sensors is more problem of get used to a determined liftoff value. This type of sensors seems to track well in a lot of surfaces).

*About your DA Chroma & Synapse 2:
DeathAdder Chroma: Pixart SDNS-3989 (~~ SDNS-3988) &  Freescale MC9S08JM32 (S08) microcontroller
Generic 3988 datasheet (Razer uses a customized version of this sensor):
https://pdf1.alldatasheet.com/datasheet-pdf/view/899014/PIXART/SDNS-3988.html

> [Razer Invicta Speed 1]
> Param0=0x18
> Param1=0x08
> Param2=0x00
> Param3=0x00
> Param4=0x10
> Param5=30
> Param6=0
> Param7=12
> Param8=12
> [Razer Invicta Speed 2]
> Param0=0x18
> Param1=0x08
> Param2=0x00
> Param3=0x00
> Param4=0x10
> Param5=30
> Param6=0
> Param7=11
> Param8=11
> [Razer Megasoma 2 1]
> Param0=0x14
> Param1=0x08
> Param2=0x00
> Param3=0x00
> Param4=0x10
> Param5=30
> Param6=0
> Param7=35
> Param8=35
> [Razer Firefly 1]
> Param0=0x18
> Param1=0x10
> Param2=0x80
> Param3=0x00
> Param4=0x10
> Param5=20
> Param6=0
> Param7=45
> Param8=44

vs Deathadder Elite: Pixart 3389 sensor and NXP LPC11U35F microcontroller & Synapse 3:

> [Razer Invicta Speed 1]
> Param0=0x0A  (Rawdata_Threshold (Param0) (0x0A) (0x3F))
> Param1=0x10  (Min_SQ_Run (Param1) (0x10) (0xFF))
> Param2=0x83
> Param3=0x00
> Param4=0x00
> Param5=00
> Param6=0
> Param7=64  (Liftoff offset (Param7))
> Param8=0
> [Razer Invicta Speed 2]
> Param0=0x0A
> Param1=0x10
> Param2=0x83
> Param3=0x00
> Param4=0x00
> Param5=00
> Param6=0
> Param7=59
> Param8=0
> [Razer XXXXX 1]
> Param0=0x0A
> Param1=0x10
> Param2=0x83
> Param3=0x00
> Param4=0x00
> Param5=00
> Param6=0
> Param7=ZZ
> Param8=0

The format is the same but is difficult to know what is what, 'cause the generic sensor datasheet shows that we barelly can write any register related with calib/liftoff (ManCal_Configuration)(VFR_Shutter_* ?¿?¿?). Anyway really this is just a mere curiosity as we can replicate the same behaviour of synapse software. And we can asume that as the format of the data that synapse sends to the microcontroller is similar, probably the functions that will be implemented for 5g infrared sensors (DAE & similar...MambaElite, NagaTrinity,etc)  will work with 4g infrared sensors (pixart s398X) but i think we need pcaps to really check it. 

Regards.

leimh, 2020-06-30:
Hi again! :)
Currently i havent got the DA Elite anymore (RMA problem) but i got the DA V2 instead ;) And well i just did a fast research about the calibration stuff in this device ("new" sensor) that maybe someone will find interesting ;P
************************
DA V2: Pixart PAW3399 sensor and NXP LPC51U68 JBD48 microcontroller.
(Other mice  with this sensor are: Viper Ultimate, Basilisk Ultimate, Basilisk V2)

The new sensor (called **FOCUS+** by Razer)  got new calibration features like:

> 
> **SMART TRACKING**
> 
> For optimal gaming precision, older sensors need to be manually calibrated each time you switch to a new mouse surface. If they aren’t tuned, this causes the lift-off distance—the distance at which the mouse stops tracking when picked up—to be inconsistent, which affects your accuracy in-game.
> 
> With Smart Tracking, the Razer™ Focus+ Optical Sensor is able to automatically calibrate itself across different mouse surfaces, ensuring your lift-off distance—and accuracy—remains consistent.

and

> **ASYMMETRIC CUT-OFF**
> 
> On top of the sensor’s lift-off distance, you can add another level of precision by using Asymmetric Cut-off to set its landing distance—the distance at which the mouse resumes tracking again. Ideally, you'd want this to be as low as possible so that less cursor drift occurs as your mouse touches the surface.
> 
[image]
*********************************
*********************************
PCAPs:

```
Status Trans Packet Proto DataSize Class CMD Args : 1_from_custom_calib_to_without_calib.pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 2_from_smart_calib_to_without_calib.pcapng ??????? TODO: A bug in synapse when asymtric was previously set ?????? but "406" new mode? 
00     1f    0000   00    03       0b    03  000401 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040600 | UNKNOWN
00     1f    0000   00    07       0b    0c  00041c300c0c0a | UNKNOWN
00     1f    0000   00    0a       0b    0d  00040e381b881b000e08 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 3_smart_calib_3mm.pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040102 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040102 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040102 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 4_smart_calib_2mm.pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x6
00     1f    0000   00    04       0b    0b  00040101 | UNKNOWN x3
Status Trans Packet Proto DataSize Class CMD Args : 5_smart_calib_1mm.pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x2
00     1f    0000   00    04       0b    0b  00040100 | UNKNOWN
(0.9 sec later) ?????? check!!
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x2
00     1f    0000   00    04       0b    0b  00040100 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 6_smart_asym_3_(up_3_down_2).pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x5
00     1f    0000   00    04       0b    0b  00040202 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040202 | UNKNOWN x2
Status Trans Packet Proto DataSize Class CMD Args : 7_smart_asym_2_(up_3_down_1).pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x6
00     1f    0000   00    04       0b    0b  00040201 | UNKNOWN x3
Status Trans Packet Proto DataSize Class CMD Args : 8_smart_asym_1_(up_2_down_1).pcapng
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x2
00     1f    0000   00    04       0b    0b  00040200 | UNKNOWN
00     1f    0000   00    03       0b    03  000400 | UNKNOWN x2
00     1f    0000   00    04       0b    0b  00040200 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 10_gigantus_lvl_5.pcapng
00     1f    0000   00    03       0b    03  000401 | UNKNOWN x2
00     1f    0000   00    04       0b    0b  00040300 | UNKNOWN
00     1f    0000   00    0a       0b    05  00042030110811050f0a | UNKNOWN
[... 98 lines cut ...]
00     1f    0000   00    04       0b    0b  00040500 | UNKNOWN
00     1f    0000   00    0a       0b    05  0004253003080a050f0f | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 22_set_custom_at_lvl_10.pcapng
00     1f    0000   00    03       0b    03  000401 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040500 | UNKNOWN
00     1f    0000   00    0a       0b    05  00040a381e081e000e08 | UNKNOWN
00     1f    0000   00    03       0b    03  000401 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040500 | UNKNOWN
00     1f    0000   00    0a       0b    05  00040a381e081e000e08 | UNKNOWN
Status Trans Packet Proto DataSize Class CMD Args : 23_set_custom_asym_lvlup_4_lvldown_1.pcapng
00     1f    0000   00    03       0b    03  000401 | UNKNOWN
00     1f    0000   00    04       0b    0b  00040600 | UNKNOWN
00     1f    0000   00    07       0b    0c  00042530030a0f | UNKNOWN lvldown ??
[... cut ...]
00     1f    0000   00    07       0b    0c  00042530030a0f | UNKNOWN lvldown ?? x3
00     1f    0000   00    0a       0b    0d  00040e381b881b000e08 | UNKNOWN lvlup?? x3
Status Trans Packet Proto DataSize Class CMD Args : 25_set_custom_asym_lvlup_9_lvldown_4.pcapng
00     1f    0000   00    03       0b    03  000401 | UNKNOWN x3
00     1f    0000   00    04       0b    0b  00040600 | UNKNOWN x3
00     1f    0000   00    07       0b    0c  00041c300c0c0a | UNKNOWN lvldown ?? x3
00     1f    0000   00    0a       0b    0d  00040e381b881b000e08 | UNKNOWN lvlup?? x3
```

leimh, 2020-06-30:
***************************************************
**mmdatav4.ini**:

> [LCO01]
> Param0=0x34
> Param1=0x30
> Param2=0x03
> Param3=0x08
> Param4=0x0A
> Param5=0x05
> Param6=0x0F
> Param7=0x0F
> [LCO02]
> Param0=0x2F
> Param1=0x30
> Param2=0x08
> Param3=0x08
> Param4=0x0A
> Param5=0x05
> Param6=0x0F
> Param7=0x0D
> ...
> [LCO10]
> Param0=0x07
> Param1=0x38
> Param2=0x20
> Param3=0x08
> Param4=0x20
> Param5=0x00
> Param6=0x0E
> Param7=0x08
> [LTT01]
> Param0=0x34 
> Param1=0x30 
> Param2=0x03 
> Param3=0x0A 
> Param4=0x0F
> ...
> [LTT10]
> Param0=0x07 
> Param1=0x38 
> Param2=0x20 
> Param3=0x20	
> Param4=0x08
> [TTL01]
> Param0=0x34 
> Param1=0x30 
> Param2=0x03 
> Param3=0x88 
> Param4=0x0A 
> Param5=0x0F 
> Param6=0x0F 
> Param7=0x0F
> ...
> [TTL10]
> Param0=0x07 
> Param1=0x38 
> Param2=0x20 
> Param3=0x88 
> Param4=0x20 
> Param5=0x00 
> Param6=0x0E 
> Param7=0x08
> [SURFLIST]
> Razer Invicta Speed
> Razer Invicta Control
> Razer Vespula Speed
> Razer Vespula Control
> Razer Vespula V2 (Hard)
> Razer Vespula V2 (Cloth)
> Razer Goliathus Control
> Razer Goliathus Speed
> Razer Kabuto
> Razer Destructor 2
> Razer Megasoma 2
> Razer Firefly Hard Edition
> Razer Sphex
> Razer Manticor
> Razer Sphex V2
> Razer Goliathus Mobile
> Razer Firefly Cloth Edition
> Razer Gigantus
> Razer Goliathus Chroma
> Razer Firefly V2

***************************************************

LCO01-LCO10:param0-param7 | normal ("symmetric") lift off values.

LTT01-LTT10:param0-param4 | asymmetric level down
TTL01-TTL10:param0-param7 | asymmetric level up

Examples:
10-gigantus level 5 ---→ LCO05
11-gigantus asymmetric lvl up 4 lvl down 1 --→ LTT01, TTL04
12-destructor 2 lvl 10 --→ LCO10
13-destructor 2 lvl 5 --→  LCO05
14-destructor 2 lvl 1 ---→ LCO01
15_destructor2_asym_lvl_up_5_lvl_down_3  ---→ LTT05, TTL03
18_destructor2_asym_lvl_up_10_lvl_down_8 ---→ LTT10, TTL08

??Example of hypothetical mousemat (TODO: need check all razer mm via pcaps)???:
xxxxxx level 10 ----→ LCO07
…
xxxxxx level 5 -----→ LCO04
xxxxxx level 4 -----→ LCO04
…
xxxxxx level 1 -----→ LCO02

*************************
**SensorCalibration_Pixart.xml (current values)**
*************************

```
...
<Mode>SmartTracking</Mode>
  <SmartTracking>
    <IsHysteresis>true</IsHysteresis>
    <TrackingDistance>2</TrackingDistance>
    <LiftOffDistance>2</LiftOffDistance>
    <LandingDistance>1</LandingDistance>
  </SmartTracking>
  <SurfaceTracking>
    <SurfaceDistance>5</SurfaceDistance>
    <LiftOffDistance>9</LiftOffDistance>
    <LandingDistance>4</LandingDistance>
  </SurfaceTracking>
  <LiftOff>3</LiftOff>
</SensorCalibration>
```

*******************************
**PixartCalibrationData.xml (for previous "12_make_custom_lvl_5.pcapng")**
*******************************

```
…
<SensorData>
        <Param0>37</Param0>
        <Param1>3</Param1>
      </SensorData>
    </PixartCalibrationData>
  </CustomCalibrationData>
</PixartCalibration>

```
**************************************

02     1f    0000   00    0b       0b    85  0004250300000000000000 | Mouse sends the params to synapse

param0: 25->37
param1: 03->3

param0 and param1 are stored in decimal notation in PixartCalibrationData.xml

***************************************
 **As resume:**
***************************************
Status Trans Packet Proto DataSize Class CMD Args : 
00     1f    0000   00    03       0b    03  000400 | Reset/Disable calibration

00     1f    0000   00    03       0b    03  000400 | 
00     1f    0000   00    04       0b    0b  0004010X | Set standard smart calibration X=(0 to 2) (hysteresis true ??? TODO)

00     1f    0000   00    03       0b    03  000400 | 
00     1f    0000   00    04       0b    0b  0004020Y | Set asymetric smart calibration Y=(0 to 2) (hysteresis false ??? TODO)

00     1f    0000   00    03       0b    03  000401 | Set predefined calibration profile in standard
00     1f    0000   00    04       0b    0b  00040300 | mode for a specific Razer mousemat.
00     1f    0000   00    0a       0b    05  0004Z | Z=(param0, param1,..., param7) in mmvdata4.ini (LCOXX values)

00     1f    0000   00    03       0b    03  000401 | Set predefined calibration profile in asymetric
00     1f    0000   00    04       0b    0b  00040400 | mode for a specific Razer mousemat.
00     1f    0000   00    07       0b    0c  0004T | Set LandingDistance T=(param0,...,param4) mmvdata4.ini (LTTXX values)
00     1f    0000   00    0a       0b    0d  0004S | Set LiftOffDistance S=(param0,..., param7) in mmvdata4.ini (TTLXX values)

Regards.

ByteDrummer, 2022-05-31:
Any progress on this? I recently bought new skates for my mouse (Razer Viper) and found that the sensor wasn't detecting the surface. Calibrating the surface on Windows sets the correct lift-off distance, but these settings aren't saved onto my mouse's memory.

Whayme, 2022-11-02:
> Any progress on this? I recently bought new skates for my mouse (Razer Viper) and found that the sensor wasn't detecting the surface. Calibrating the surface on Windows sets the correct lift-off distance, but these settings aren't saved onto my mouse's memory.

Also isn't saved to onboard memory on the DAV2 for some bizarre reason.

geezmolycos, 2024-10-24:
I made a document of my analysis and a web-app for Basilisk V3 liftoff and calibration features. The method to set lift-off distance, initialize a calibration and algorithm to calculate paramaters is thoroughly analyzed and implemented in the web-app. However, it's for Basilisk V3 only. That model uses PAW3399 sensor and LPC51U68 microcontroller. The protocol is the same mentioned by @leimh , it seems that the protocol can be adapted to other models with this sensor.

The document link is <https://github.com/geezmolycos/razerqdhid/blob/main/docs/cmd_sensor-en.md>

YoJames2019, 2024-10-29:
any progress on this?