openrazer/openrazer #788: Error: retrieving gpg key timed out.

spflanze, 2019-03-14:
### Distribution

### Support

- [Y] Have you tried rebooting?
- [Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My OS is Ubuntu 18.04. When I attempt to install the ppa on my system with the command:
`sudo add-apt-repository ppa:openrazer/stable`
Which is provided on page: https://openrazer.github.io/
I get this error:
> Error: retrieving gpg key timed out.

z3ntu, 2019-03-14:
The ubuntu keyserver (http://keyserver.ubuntu.com) seems to be down at the moment, which could be the origin of your troubles - or your network blocks PGP keyservers.

You can try running `sudo apt-key adv --keyserver pgp.surfnet.nl --recv-keys 903936CAB6049E2E6C33D5D8073E051D7B2AEE37` which might work (the keyserver is working on my end and the long string is the signing key used by launchpad for this PPA)

closed 2019-03-14 by z3ntu

spflanze, 2019-03-15:
Thank you for your reply. It is possible the necessary port to get the gpg key is closed in my router. Is there a port number associated with gpg keys?

Your command to install the PPA that includes the signing key worked. But my Razer Kraken 7.1 Chroma V2 is not working. Attached is the output of the command: lsub -v
[lsusb Info.txt](https://github.com/openrazer/openrazer/files/2969003/lsusb.Info.txt)
The system knows it is there and can identify it. But it cannot be opened.

z3ntu, 2019-03-15:
If the manual command worked, its very likely that it was really the Ubuntu keyserver being down was the problem.

For your current problem you will likely find the solution on the troubleshooting page linked in the first comment

spflanze, 2019-03-15:
I followed directions as well as I could at: https://github.com/openrazer/openrazer/wiki/Troubleshooting

I added my user name to the plugdev group. But the "lsusb -v" command shows the headphones still could not be opened and I do not hear anything in the headphones.

There is not an "openrazer" directory in the path `~/.local/share/`.

The command:
`sudo journalctl -b | grep -i razer_mount`
does not output anything.

When I execute the commnad:
`openrazer-daemon -Fv`
I see these lines in its output:
```
2019-03-15 12:23:49 | razer                          | DEBUG    | Loaded device specification: RazerKraken71 --------------------------- (1532:0501)
2019-03-15 12:23:49 | razer                          | DEBUG    | Loaded device specification: RazerKraken71Chroma --------------------- (1532:0504)
2019-03-15 12:23:49 | razer                          | DEBUG    | Loaded device specification: RazerKraken71V2 ------------------------- (1532:0510)
```

z3ntu, 2019-03-15:
So `sudo dkms install openrazer-driver/2.4.0` succeeds?
What about `sudo modprobe razerkbd`?

spflanze, 2019-03-16:
```
$ sudo dkms install openrazer-driver/2.4.0
Module openrazer-driver/2.4.0 already installed on kernel 4.15.0-46-lowlatency/x86_64
```
I did not execute `sudo modprobe razerkbd` because it is not a keyboard I need to get working. I nevertheless did it to answer your question. The command did not send any output. I still do not hear anything. and the command 'lsusb -v' still shows the Razer device cannot be opened.

z3ntu, 2019-03-16:
>  "lsusb -v" still showed the device could not be opened

This has no significance at all. You can run lsusb as root and it won't show that message.

Also your device (1532:0502) just is not supported, that's why it can't be found by OpenRazer...