openrazer/openrazer #790: Support for [Naga V2 2014]

gawronix, 2019-03-19:
### Device Info

* Name: <Razer Naga Hex v2 2014>
* Product Number: <RZ01-01600100>
* Model Number: <RZ01-0160>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathStalker Essential 2014 Consumer Control	id=10	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathStalker Essential 2014 	id=12	[slave  pointer  (2)]
⎜   ↳ Razer Razer Naga Hex V2                 	id=13	[slave  pointer  (2)]
⎜   ↳ Razer Razer Naga Hex V2                 	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Razer Razer DeathStalker Essential 2014 	id=8	[slave  keyboard (3)]
    ↳ Razer Razer DeathStalker Essential 2014 Keyboard	id=9	[slave  keyboard (3)]
    ↳ Razer Razer DeathStalker Essential 2014 System Control	id=11	[slave  keyboard (3)]
    ↳ Razer Razer Naga Hex V2                 	id=15	[slave  keyboard (3)]
    ↳ Razer Razer DeathStalker Essential 2014 Consumer Control	id=16	[slave  keyboard (3)]
    ↳ Razer Razer Naga Hex V2                 	id=17	[slave  keyboard (3)]

```
Please post the output of
    ls -lah /dev/input/by-id/
```
razem 0
drwxr-xr-x 2 root root 220 mar 19 21:37 .
drwxr-xr-x 4 root root 540 mar 19 21:37 ..
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_DeathStalker_Essential_2014-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_DeathStalker_Essential_2014-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_DeathStalker_Essential_2014-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_DeathStalker_Essential_2014-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_DeathStalker_Essential_2014-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_Naga_Hex_V2-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_Naga_Hex_V2-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 mar 19 21:37 usb-Razer_Razer_Naga_Hex_V2-if02-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 mar 19 21:37 usb-Razer_Razer_Naga_Hex_V2-mouse -> ../mouse1

### Packet Captures
This is side buttons from xev

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3344677, (383,207), root:(2408,758),
    state 0x10, keycode 10 (keysym 0x31, 1), same_screen YES,
    XLookupString gives 1 bytes: (31) "1"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3345206, (383,207), root:(2408,758),
    state 0x10, keycode 11 (keysym 0x32, 2), same_screen YES,
    XLookupString gives 1 bytes: (32) "2"
    XmbLookupString gives 1 bytes: (32) "2"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3345329, (383,207), root:(2408,758),
    state 0x10, keycode 11 (keysym 0x32, 2), same_screen YES,
    XLookupString gives 1 bytes: (32) "2"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3346282, (383,207), root:(2408,758),
    state 0x10, keycode 12 (keysym 0x33, 3), same_screen YES,
    XLookupString gives 1 bytes: (33) "3"
    XmbLookupString gives 1 bytes: (33) "3"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3346423, (383,207), root:(2408,758),
    state 0x10, keycode 12 (keysym 0x33, 3), same_screen YES,
    XLookupString gives 1 bytes: (33) "3"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3346773, (383,207), root:(2408,758),
    state 0x10, keycode 13 (keysym 0x34, 4), same_screen YES,
    XLookupString gives 1 bytes: (34) "4"
    XmbLookupString gives 1 bytes: (34) "4"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3346929, (383,207), root:(2408,758),
    state 0x10, keycode 13 (keysym 0x34, 4), same_screen YES,
    XLookupString gives 1 bytes: (34) "4"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3347273, (383,207), root:(2408,758),
    state 0x10, keycode 14 (keysym 0x35, 5), same_screen YES,
    XLookupString gives 1 bytes: (35) "5"
    XmbLookupString gives 1 bytes: (35) "5"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3347429, (383,207), root:(2408,758),
    state 0x10, keycode 14 (keysym 0x35, 5), same_screen YES,
    XLookupString gives 1 bytes: (35) "5"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3347776, (383,207), root:(2408,758),
    state 0x10, keycode 15 (keysym 0x36, 6), same_screen YES,
    XLookupString gives 1 bytes: (36) "6"
    XmbLookupString gives 1 bytes: (36) "6"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3347900, (383,207), root:(2408,758),
    state 0x10, keycode 15 (keysym 0x36, 6), same_screen YES,
    XLookupString gives 1 bytes: (36) "6"
    XFilterEvent returns: False

KeyPress event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3348230, (383,207), root:(2408,758),
    state 0x10, keycode 16 (keysym 0x37, 7), same_screen YES,
    XLookupString gives 1 bytes: (37) "7"
    XmbLookupString gives 1 bytes: (37) "7"
    XFilterEvent returns: False

KeyRelease event, serial 40, synthetic NO, window 0x2a00001,
    root 0x4d9, subw 0x0, time 3348347, (383,207), root:(2408,758),
    state 0x10, keycode 16 (keysym 0x37, 7), same_screen YES,
    XLookupString gives 1 bytes: (37) "7"
    XFilterEvent returns: False

### Synapse Screenshots
https://pasteboard.co/I6cpSSb.png
https://pasteboard.co/I6cqaBf.png
https://pasteboard.co/I6cqixf.png
https://pasteboard.co/I6cqoWA.png
https://pasteboard.co/I6cquC8t.png
https://pasteboard.co/I6cqGTU.png

z3ntu, 2019-03-20:
The device Razer Naga Hex V2 (1532:0050) is already supported. If your device has a different USB ID, please post the lsusb output as requested in the issue template.

gawronix, 2019-03-20:
> The device Razer Naga Hex V2 (1532:0050) is already supported. If your device has a different USB ID, please post the lsusb output as requested in the issue template.

Bus 008 Device 002: ID 1532:0050 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0050 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
[... 144 lines cut ...]
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

gawronix, 2019-03-20:
i cant use my side keys maped with numlock pad from keyboard. It not working in dota 2 and other programs , keys is normal 1,2,3,4,5,6,7 not Num1,Num2 etc.

z3ntu, 2019-03-20:
Key remapping is not supported (and won't be in the near future). There are other software solutions for that out there though.

closed 2019-03-20 by z3ntu

gawronix, 2019-03-20:
can you sent me link to this software because i cant find any...