openrazer/openrazer #800: Huntsman Elite is still not supported [2.5.0

BERNARDO31P, 2019-03-28:
### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

z3ntu, 2019-03-28:
`sudo modprobe razerkbd` works? Is there the "RazerHuntsmanElite" string in the daemon output?

BERNARDO31P, 2019-03-28:
`sudo modprobe razerkbd` does work, there's no output.

z3ntu, 2019-03-29:
Something at `sudo journalctl -b | grep -i razer_mount`?

kalanihelekunihi, 2019-03-30:
I needed to reboot after running modprobe razerkbd prior to the Huntsman Elite showing up. Prior to that point, I saw similar behavior to what Bernardo described.

z3ntu, 2019-03-30:
OP has tried rebooting according to the first comment.

@kalanihelekunihi So the Huntsman Elite works for you on 2.5.0?

kalanihelekunihi, 2019-03-30:
Yes. running: openrazer-daemon 2.5.0 and RazerGenie 0.8.1
Initially, the keyboard was not detected, however after running modprobe razerkbd, it was and is properly running synced effects with my Mamba Tournament Edition and Firefly.

Device is also listed under journalctl: 
Mar 30 14:03:51 lorelei-desktop razer_mount[1433]: Driver razerkbd
Mar 30 14:03:51 lorelei-desktop razer_mount[1435]: Driver razerkbd
Mar 30 14:03:51 lorelei-desktop razer_mount[1434]: Driver razerkbd
Mar 30 14:03:51 lorelei-desktop razer_mount[1436]: Device_ID 0003:1532:0226.0006
Mar 30 14:03:51 lorelei-desktop razer_mount[1437]: Device_ID 0003:1532:0226.0008
Mar 30 14:03:51 lorelei-desktop razer_mount[1438]: Device_ID 0003:1532:0226.0003
Mar 30 14:03:51 lorelei-desktop razer_mount[1439]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0008/
Mar 30 14:03:51 lorelei-desktop razer_mount[1440]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0003/
Mar 30 14:03:51 lorelei-desktop razer_mount[1441]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0006/
Mar 30 14:03:51 lorelei-desktop razer_mount[1445]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0006/
Mar 30 14:03:51 lorelei-desktop razer_mount[1446]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0003/
Mar 30 14:03:51 lorelei-desktop razer_mount[1447]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:0226.0008/
Mar 30 14:03:51 lorelei-desktop razer_mount[1459]: Driver razermouse
Mar 30 14:03:51 lorelei-desktop razer_mount[1461]: Driver razermouse
Mar 30 14:03:51 lorelei-desktop razer_mount[1460]: Driver razermouse
Mar 30 14:03:51 lorelei-desktop razer_mount[1462]: Device_ID 0003:1532:0046.0007
Mar 30 14:03:51 lorelei-desktop razer_mount[1463]: Device_ID 0003:1532:0046.0005
Mar 30 14:03:51 lorelei-desktop razer_mount[1464]: Device_ID 0003:1532:0046.0009
Mar 30 14:03:51 lorelei-desktop razer_mount[1465]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0007/
Mar 30 14:03:51 lorelei-desktop razer_mount[1467]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0005/
Mar 30 14:03:51 lorelei-desktop razer_mount[1466]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0009/
Mar 30 14:03:51 lorelei-desktop razer_mount[1471]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0007/
Mar 30 14:03:51 lorelei-desktop razer_mount[1472]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0005/
Mar 30 14:03:51 lorelei-desktop razer_mount[1473]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0046.0009/
Mar 30 14:03:51 lorelei-desktop razer_mount[1493]: Driver razermousemat
Mar 30 14:03:51 lorelei-desktop razer_mount[1495]: Driver razermousemat
Mar 30 14:03:51 lorelei-desktop razer_mount[1494]: Driver razermousemat
Mar 30 14:03:51 lorelei-desktop razer_mount[1496]: Device_ID 0003:1532:0C00.000A
Mar 30 14:03:51 lorelei-desktop razer_mount[1497]: Device_ID 0003:1532:0C00.000C
Mar 30 14:03:51 lorelei-desktop razer_mount[1498]: Device_ID 0003:1532:0C00.000B
Mar 30 14:03:51 lorelei-desktop razer_mount[1499]: Changing group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000A/
Mar 30 14:03:51 lorelei-desktop razer_mount[1501]: Changing group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000B/
Mar 30 14:03:51 lorelei-desktop razer_mount[1500]: Changing group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000C/
Mar 30 14:03:51 lorelei-desktop razer_mount[1505]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000B/
Mar 30 14:03:51 lorelei-desktop razer_mount[1506]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000A/
Mar 30 14:03:51 lorelei-desktop razer_mount[1507]: Changed group /sys/bus/hid/drivers/razermousemat/0003:1532:0C00.000C/

z3ntu, 2019-03-30:
Great, thanks for the confirmation.

Rebooting is "needed" (there are other ways too but this one is the easiest) to trigger the udev rule to rebind the device from generic hid to our driver

closed 2019-03-30 by z3ntu

BERNARDO31P, 2019-04-02:
I tried everything and it still doesn't work.

z3ntu, 2019-04-03:
Can you run the following commands and post the output here?

`modinfo razerkbd`

and

`ls -al /sys/bus/hid/drivers/razer*/0*/`

BERNARDO31P, 2019-04-03:
Here's the output from the first command [modinfo](https://gist.github.com/BERNARDO31P/eff3a0066ae9c7515165529e3a59983c) and here's the output of the second command 
```
[bernardo@archdesk ~]$ ls -al /sys/bus/hid/drivers/razer*/0*/
ls: Zugriff auf '/sys/bus/hid/drivers/razer*/0*/' nicht möglich: Datei oder Verzeichnis nicht gefunden
[bernardo@archdesk ~]$ sudo ls -al /sys/bus/hid/drivers/razer*/0*/
[sudo] Passwort für bernardo: 
ls: Zugriff auf '/sys/bus/hid/drivers/razer*/0*/' nicht möglich: Datei oder Verzeichnis nicht gefunden
```
and that translated means "`Access to ### not possible: File or directory not found`"

z3ntu, 2019-04-03:
What about `sudo journalctl -b | grep -i razer_mount` ?

BERNARDO31P, 2019-04-03:
```
bernardo@archdesk ~> sudo journalctl -b | grep -i razer_mount 
[sudo] Passwort für bernardo: 
bernardo@archdesk ~> 
```

z3ntu, 2019-04-03:
`lsusb | grep 1532` ?

z3ntu, 2019-04-03:
`1532:0228` is not yet supported, that's why nothing happens.

z3ntu, 2019-04-03:
And `1532:0228` is a Razer BlackWidow Elite according to my spreadsheet, are you sure it's a Huntsman Elite?

BERNARDO31P, 2019-04-03:
Oh yeah I see, it's the BlackWidow Elite, when will it be supported?

z3ntu, 2019-04-03:
That is issue #684