openrazer/openrazer #802: no results found for openrazer

DatOneLefty, 2019-03-29:
i ran `pacaur -S openrazer-meta` and `env LC_ALL=C pacaur -S openrazer-meta` and got the following output:
```:: Package openrazer-meta not found in repositories, trying AUR...
:: resolving dependencies...
:: looking for inter-conflicts...

AUR Packages  (5) openrazer-daemon-2.5.0-1  openrazer-driver-dkms-2.5.0-1  openrazer-meta-2.5.0-1  python-notify2-0.3.1-2  python-openrazer-2.5.0-1  
Repo Packages (8) cblas-3.8.0-2  dkms-2.5-4  python-daemonize-2.5.0-1  python-numpy-1.16.2-1  python-pyudev-0.21.0-4  python-setproctitle-1.1.10-3  xautomation-1.09-3  xdotool-3.20160805.1-2  

Repo Download Size:    2.89 MiB
Repo Installed Size:  19.12 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving package(s)...
no results found for openrazer
clone complete: /home/arch/.cache/pacaur/python-notify2
:: failed to retrieve openrazer package
```

i also tried using the tar.gz from the AUR site and it gave me the following error when running `pacman -U *xz`:
```
loading packages...
error: missing package metadata in openrazer-2.5.0.tar.xz
error: 'openrazer-2.5.0.tar.xz': invalid or corrupted package
```

Is installation broken or am I installing it wrong?

z3ntu, 2019-03-29:
That's a bug in the "new" pacaur (since it was forked from the original author). Just follow https://wiki.archlinux.org/index.php/Arch_User_Repository#Installing_packages and you should be good. 

You might want to install `python-notify2` with pacaur first though as it's an AUR-only dependency, that you would have to build manually too.

z3ntu, 2019-07-12:
Closing as there was no response.

closed 2019-07-12 by z3ntu