openrazer/openrazer #803: Razer Huntsman Elite - Volume Knob is a Scroll Wheel

kalanihelekunihi, 2019-03-30:
### Distribution
Ubuntu 18.04

### Support
- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

When using the openrazer-daemon 2.5.0, after running modprobe razerkbd, the volume knob on the keyboard is remapped to a scroll wheel. On unloading the module and rebooting, the volume knob returns to it's normal function.

Please advise if there are any logs or details that I can provide to replicate this behavior.

z3ntu, 2019-04-01:
Can you try stopping the daemon (`systemctl --user stop openrazer-daemon.service`) and report back if the behavior changes?

BDiamond, 2019-04-13:
I have the same issue. The Knob acts as a scroll wheel when the daemon is running and returns to its normal volume control behavior when the daemon is stopped.

beranat, 2019-04-15:
Dear **@BDiamond** and  **@kalanihelekunihi**, do Multimedia Buttons (Next Song, Play/Pause and Prev Song) work?

BDiamond, 2019-04-15:
Nope, they don't work when the daemon is running.

beranat, 2019-04-15:
Dear **@BDiamond**, would like to thank you and offer to check gittag#d48ea90 [Add Razer BlackWidow Elite support #804 ](https://github.com/openrazer/openrazer/pull/804)

andrewschott, 2019-04-27:
I am having the same issue on OpenSUSE Tumbleweed.  The daemon running remaps the volume wheel to a scroll, and the multimedia buttons do not work at all.

micach, 2019-05-23:
I've fixed this.
You need to use @beranat 's  branch:
https://github.com/beranat/openrazer.git
And add cases for the Huntsman (USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE) elite as it is added for the Blackwidow elite.
and remove this:
`    // ...and on these keyboards`
`    switch (usb_dev->descriptor.idProduct) {`
`    case USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE:`
`        return 0;`
`    }`
in _razerkbd_driver.c_
After recompilation, all works like a charm. Lightning, volume knob and media keys..

rgarc, 2019-05-25:
> And add/change cases for the Huntsman (USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE) elite as it is added for the USB_DEVICE_ID_RAZER_BLACKWIDOW_ELITE.
> I mean if somewhere USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE is used in a different section it should be moved to the same section as USB_DEVICE_ID_RAZER_BLACKWIDOW_ELITE

Dear @micach can you explain this part in more detail? I've not familiar w/ the codebase and was only able to comment out the part you said in `razerkbd_driver.c`

micach, 2019-05-25:
You need to remove existing cases for USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE
`case USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE:`
And add it above every
`case USB_DEVICE_ID_RAZER_BLACKWIDOW_ELITE:`

On mentioned branch.

beranat, 2019-05-25:
Dear [b]@micach[b] and [b]@rgarc[/b], I would thank you for help.

[b]@micach[/b] - I not clearly understood - should be light effects block corrected (or only keyboard remapping code)?

[b]rgarc[/b] I have fixed code [during this topic](https://github.com/openrazer/openrazer/issues/803#issuecomment-495198136)

micach, 2019-05-25:
I've changed every single place in `razerkbd_driver.c`. Lightning too. But it was working before. Anyway, the only difference is 
report.transaction_id.id and seems it doesn't matter(?). I should check it on Wireshark.

And one more place to be corrected:
`// The volume scroller becomes mouse's scroll for USB_DEVICE_ID_RAZER_BLACKWIDOW_ELITE:`
` if ((usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_ELITE`
`    || usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_HUNTSMAN_ELITE)`
`    && intf->cur_altsetting->desc.bInterfaceProtocol == USB_INTERFACE_PROTOCOL_MOUSE`
`    && size == 8 && data[0] == 0x00) {`

beranat, 2019-05-25:
This code block is obsoleted - it's functionality is transformed into "input mapping" (function razer_kbd_input_mapping).

micach, 2019-05-25:
@beranat yes, as I said it change nothing. I've simplified change details because of this. To achive media buttons and volume knob handling for huntsman elite, if mentioned code Block is obsolete, the only crucial change is mapping keys same way as it has been done for black widow elite.

andrewschott, 2019-05-26:
> **@micach**, I not clearly understood - should be light effects block corrected (or only keyboard remapping code)?

The lighting is perfect with the master branch here, openrazer official github page, (well, at least the Fedora and OpenSUSE packages are).  The _ONLY_ problem is the wheel is showing up as a mouse scroll wheel.  Otherwise the Huntsman Elite is perfect (actually was a few releases back).

micach, 2019-05-26:
@andrewschott there is also problem with media buttons. Replace your razerkbd_driver.c with one I've provided above, recompile and all is working well. After new release, it will be probably included in main branch (after alignment discussed with @beranat)

danisawesome, 2019-07-03:
Hey all! I wanted to say thanks for the discussion as I was able to get my Huntsman Elite's media buttons working by forking @beranat 's fork and applying @micach 's instructions. I also was able to get underglow working which was the last piece that was bugging me (only 6 rows were specified in the daemon for the Huntsman Elite when really there are 9).

The media key and underglow fixes for the Huntsman Elite are in my fork here: https://github.com/danisawesome/openrazer

commit 03a48901 (2019-07-13): Add Razer BlackWindow Elite support

Fixes #684, fixes #803
closed 2019-07-13 by z3ntu