openrazer/openrazer #807: Docker Openrazer + polychromatic

Felipefuji, 2019-04-01:
First, apologize if what I say is crazy. Second, would it be possible to create a docker with openrazer and polychromatic? it occurred to me to have it run the Mac OS, and to be able to have gui, it can be done with socat. If you think it is possible, I will try to work on it, or if someone has proven it to inform me. Thank you very much for your time and dedication.

dimyme, 2019-04-01:
what's the advantage of that?

Felipefuji, 2019-04-01:
to be able to use it on Mac osx without you having to do any recoding. If possible, I still do not know.

Felipefuji, 2019-04-01:
> 
> 
> what's the advantage of that?

to be able to use it on Mac osx without you having to do any recoding. If possible, I still do not know.

z3ntu, 2019-04-01:
No, that won't work because OpenRazer uses Linux kernel modules to communicate with the Razer devices and that module won't work on macOS.

If you want to use a similar thing to OpenRazer on macOS though, you can try my https://github.com/z3ntu/razer_test project, it does work fine on macOS. Ping me on Telegram (@z3ntu), Matrix (@z3ntu:matrix.org) or IRC (z3ntu @ freenode) if you're interested.

closed 2019-04-01 by z3ntu