openrazer/openrazer #808: reopen wiki for contributors

dimyme, 2019-04-01:
seems like a critique is unwanted in this project. Why was the wiki restricted recently? No open discussion wanted?

z3ntu, 2019-04-01:
The wiki is not a place for discussion to happen, issues are better suited for that. 

And we are very much aware of the limitations of the current macro system and we are welcoming contributions on improving that.

closed 2019-04-01 by z3ntu

dimyme, 2019-04-01:
well lets discuss it right here and now. what is required to have persistent macros ? probably some UDEV rule and an actual "macro downloader" I suppose?

for a free razer wiki look at:   https://github.com/dimyme/openrazer/wiki

dimyme, 2019-04-01:
Always nice to hear that you are "welcoming contributions" at the same time as you are deleting wiki contribs.

z3ntu, 2019-04-01:
> what is required to have persistent macros ?

I don't know because I am not familiar with the macro code in OpenRazer and macros in general

> Always nice to hear that you are "welcoming contributions" at the same time as you are deleting wiki contribs.

Writing your personal experiences/annoyances into a project wiki (= documentation) is not the same as opening a pull request to improve macro support.

But just so you know, your way of interacting with other people feels very rude, so you might want to improve that if you don't want to piss people off.