openrazer/openrazer #809: Razer Blade Stealth (all versions) should support for Starlight Matrix Effect

jmachen, 2019-04-02:
This effect is confirmed supported/working in Windows on every iteration of the Razer Blade Stealth, but is not enabled in the driver.

mgruppi, 2019-04-04:
Are you currently using a Blade Stealth 2019? I believe you cannot set individual lighting either, is that correct? A quick update to the driver file should do the job but I'd like you to confirm the issue is really happening on the Blade Stealth 2019.

jmachen, 2019-04-05:
No, I'm still using an early 2016 version (Model RZ09-01682E22)

jmachen, 2019-04-05:
Package: openrazer-meta
Version: 2.5.0-1~422~ubuntu18.10.1
Priority: optional
Section: misc
Source: openrazer
Maintainer: Terry Cain <terry@terrys-home.co.uk>
Installed-Size: 30.7 kB
Depends: openrazer-driver-dkms (= 2.5.0-1~422~ubuntu18.10.1), openrazer-daemon (= 2.5.0-1~422~ubuntu18.10.1), python3-openrazer (= 2.5.0-1~422~ubuntu18.10.1)
Suggests: openrazer-doc (= 2.5.0-1~422~ubuntu18.10.1)
Download-Size: 12.9 kB
APT-Manual-Installed: yes
APT-Sources: http://ppa.launchpad.net/openrazer/daily/ubuntu cosmic/main amd64 Packages
Description: OpenRazer peripheral drivers (metapackage)
 OpenRazer is a collection of GNU/Linux drivers for the Razer devices.
 Supported devices include keyboards, mice, mouse-mats, headsets and
 various other devices.
 .
 This package is a metapackage which depends on the OpenRazer driver and
 userspace daemon, plus a Python library and documentation.

Package: openrazer-meta
Version: 2.5.0~ubuntu18.10.1
Priority: optional
Section: misc
Source: openrazer
Maintainer: Terry Cain <terry@terrys-home.co.uk>
Installed-Size: 30.7 kB
Depends: openrazer-driver-dkms (= 2.5.0~ubuntu18.10.1), openrazer-daemon (= 2.5.0~ubuntu18.10.1), python3-openrazer (= 2.5.0~ubuntu18.10.1)
Suggests: openrazer-doc (= 2.5.0~ubuntu18.10.1)
Download-Size: 12.8 kB
APT-Sources: http://ppa.launchpad.net/openrazer/stable/ubuntu cosmic/main amd64 Packages
Description: OpenRazer peripheral drivers (metapackage)
 OpenRazer is a collection of GNU/Linux drivers for the Razer devices.
 Supported devices include keyboards, mice, mouse-mats, headsets and
 various other devices.
 .
 This package is a metapackage which depends on the OpenRazer driver and
 userspace daemon, plus a Python library and documentation.

jmachen, 2019-04-05:
Also, I'm not a C/C++ developer, but this segment taken from openrazer/driver/razerkbd_driver.c would seem to support the argument that the starlight effect is disabled (by omission) at the driver level.

        case USB_DEVICE_ID_RAZER_BLADE_STEALTH:
        case USB_DEVICE_ID_RAZER_BLADE_STEALTH_LATE_2016:
        case USB_DEVICE_ID_RAZER_BLADE_STEALTH_MID_2017:
        case USB_DEVICE_ID_RAZER_BLADE_STEALTH_LATE_2017:
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_wave);            // Wave effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_spectrum);        // Spectrum effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_none);            // No effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_reactive);        // Reactive effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_breath);          // Breathing effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_static);          // Static effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_custom);          // Custom effect
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_custom_frame);           // Set LED matrix
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_led_state);                // Enable/Disable the logo
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_fn_toggle);                     // Sets whether FN is requires for F-Keys
            break;

mgruppi, 2019-04-05:
Okay, I think I've got it. I'll be opening a pull request soon with the fix.

commit 16dca370 (2019-07-12): Add starlight effects (random, single, dual) to some Blade laptops

 - Blade Stealth 2016 (early and late)
 - Blade Stealth 2017 (mid and late)
 - Blade Advanced 2019 (early and mid)

Fixes #809, closes #813
closed 2019-07-12 by z3ntu