openrazer/openrazer #810: Support for Razer BlackWidow Lite (2018) 1532:0235 (openrazer-2.5.0 patch)

mfabijanic, 2019-04-03:
### Device Info

* Name: Razer BlackWidow Lite
* Product Number: RZ03-02640300-R3W1
* Model Number: RZ03-0264

### Device Information
```
Bus 001 Device 010: ID 1532:0235 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0235 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Lite
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx 1 root root  10 tra   3 16:25 usb-Razer_Razer_BlackWidow_Lite-event-kbd -> ../event19
lrwxrwxrwx 1 root root  10 tra   3 16:25 usb-Razer_Razer_BlackWidow_Lite-if01-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 tra   3 16:25 usb-Razer_Razer_BlackWidow_Lite-if02-event-mouse -> ../event18
lrwxrwxrwx 1 root root   9 tra   3 16:25 usb-Razer_Razer_BlackWidow_Lite-if02-mouse -> ../mouse3
```

### Support for BlackWidow Lite (2018) 1532:0235 - openrazer-2.5.0 patch

[openrazer-2.5.0-blackwidow_lite_support.zip](https://github.com/openrazer/openrazer/files/3039202/openrazer-2.5.0-blackwidow_lite_support.zip)

z3ntu, 2019-04-03:
@mfabijanic Do all the effects work with openrazer with the patch?

mfabijanic, 2019-04-03:
@z3ntu It works with Breath, Static and None like in Razer Synapse 3.

z3ntu, 2019-04-06:
I've pushed your patch (with some unnecessary bits removed) to the `feature_blackwidow_lite_0x0235` branch. Could you try that quickly?

mfabijanic, 2019-04-06:
I will try that next monday. A am at moto party right now.

mfabijanic, 2019-04-08:
It works. Thank you.

commit 21dffa42 (2019-04-13): Add support for Razer BlackWidow Lite

Closes #810
closed 2019-04-13 by z3ntu