openrazer/openrazer #811: Support for CYNOSA Pro 1532:020d

mamooti, 2019-04-03:
### Device Info

* Name: CYNOSA Pro
* Product Number: RZ03-01470100
* Model Number: RZ03-0147

### Device Information
```
Bus 001 Device 010: ID 1532:020d Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x020d 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
[... 276 lines cut ...]
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

### Packet Captures
<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

FuzzyGamers, 2019-04-15:
Hey I have The Cynosa Pro on Ubuntu. How can I send you the PCAP?

z3ntu, 2019-04-16:
It's explained on this wiki page: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Once you've got the .pcapng files, pack them into a zip and drag the file into the comment box here.

FuzzyGamers, 2019-09-16:
Alight so this is going way over my head, i really hope someone sent you a pcap of it. otherwise is there a youtube video going over this by any chance?

OkamiBear, 2021-07-02:
Hello... not a dev here... but l would certainly love support added for the Razer Cynosa Pro Bundle [1532:020d: Keyboard] and [1532:004d : Mouse]. It is mentioned that "PCAPS" is needed? I'd love to help anyway i can... if i can get the PCAPS following @z3ntu 's guide... support might be added yes?

z3ntu, 2021-07-03:
@OkamiBear In short, if you want some pointers to help implement support for this device, then packet captures are needed, yes. (and in general having a packet capture for every device is quite useful for future referencing also)

PseudoDistant, 2022-03-24:
Has anyone worked on this?