openrazer/openrazer #814: In-hardware wave effect control

mgruppi, 2019-04-05:
I was wondering if it is possible to control the wave (and maybe other effects) in hardware. Razer Synapse lets you change a wave's direction and speed, which I believe are done in hardware. I wanted to mess around to see if it would be possible to do it here.

I've noticed that in `driver/razerchromacommon.c` :

```
struct razer_report razer_chroma_standard_matrix_effect_wave(unsigned char variable_storage, unsigned char led_id, unsigned char wave_direction)
{
    struct razer_report report = get_razer_report(0x03, 0x0A, 0x02);
    report.arguments[0] = 0x01; // Effect ID
    report.arguments[1] = clamp_u8(wave_direction, 0x01, 0x02);

    return report;
}
```

I've got a few questions regarding this function, I'd appreaciate if someone could clarify them to me:

- The second byte in report.arguments is clamped between 0x01 and 0x02. Is there a particular reason for that? Maybe safety issues? 
- Any chances other bytes in report.arguments may be used for speed?
- I wonder what are the chances of bricking my keyboard if I decide to mess up with those bytes :sweat_smile:

tildearrow, 2019-04-06:
The hardware effect has fixed speed. The reason why Synapse has variable speed is because the effect is no longer done in hardware, but rather, in software.

If you try to use a non-0x01-or-0x02 value for the direction, the results become unpredictable. Sometimes it doesn't do anything, and sometimes it displays a static wave.

mgruppi, 2019-04-06:
You are totally right. I guess I had the wrong hunch, Synapse totally controls the keyboard via software and, after testing it more closely I realized I was mistaken. 
Thanks for the heads up,

closed 2019-04-06 by mgruppi