openrazer/openrazer #815: Razer DeathAdder 3500 not detected

redirus, 2019-04-10:
The system is Ubuntu 18.04.2.

nickluijtgaarden, 2019-04-10:
Razer DeathAdder 3500 is also not detected on my system. Also running Ubuntu 18.04.2. I have added my user to the plugdev group.

> sudo gpasswd -a USERNAME plugdev

I also ran:

> modprobe razermouse

and

< lsmod | grep razermouse

Output:

> razermouse             65536  0
hid                   122880  4 usbhid,razerkbd,hid_generic,razermouse

I also checked the `/sys/bus/hid/drivers' folder

> ls -l /sys/bus/hid/drivers

Output:

> total 0
drwxr-xr-x 2 root root 0 apr 10 09:03 hid-generic
drwxr-xr-x 2 root root 0 apr 10 09:03 razerkbd
drwxr-xr-x 2 root root 0 apr 10 09:05 razermouse

So, the kernel driver seems to be loaded properly. Still, Polychromatic controller does not recognize the device. However, when looking at kernel logs, it seems that the Razer DeathAdder is getting detected. 

< dmesg

Output (last log on usb 3-4 port)

<  [ 2783.051706] usb 3-4: USB disconnect, device number 3
[ 3240.868169] usb 3-4: new full-speed USB device number 4 using xhci_hcd
[ 3241.019276] usb 3-4: New USB device found, idVendor=1532, idProduct=0016, bcdDevice= 1.00
[ 3241.019278] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3241.019278] usb 3-4: Product: Razer DeathAdder
[ 3241.019279] usb 3-4: Manufacturer: Razer
[ 3241.029068] input: Razer Razer DeathAdder as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1532:0016.0007/input/input22
[ 3241.088265] hid-generic 0003:1532:0016.0007: input,hidraw3: USB HID v1.11 Mouse [Razer Razer DeathAdder] on usb-0000:00:14.0-4/input0

Anyone got any thoughts on this?

mgruppi, 2019-04-10:
One thing that stands out to me is the PID of your product. Your idProduct (PID) is 0016 and the one listed on the webpage is 0054.
In `driver/razermouse_driver.h.36`:
`#define USB_DEVICE_ID_RAZER_DEATHADDER_3500 0x0054`

You could try changing it to 0x0016, recompiling and reinstalling the driver.

z3ntu, 2019-04-13:
Duplicate of #231

closed 2019-04-13 by z3ntu