openrazer/openrazer #816: I want to support you!

auipga, 2019-04-11:
I want to
a) support you ( @lah7 @terrycain @z3ntu )
b) indicate, which issues are important to me (with some beers)

Please register this repository in bountysource.com, issuehunt.io or similar.

z3ntu, 2019-04-13:
I don't think throwing small amounts of money at us will help much (although we appreciate it) :wink: 

We don't have unlimited amounts of time but Razer seems to have unlimited devices, so code contributions are what's helping us most.

If you really want to donate, Terry's PayPal email is in the README and for me it's https://www.paypal.me/z3ntu

Thanks!

closed 2019-04-13 by z3ntu

lah7, 2019-04-13:
As @z3ntu said, thanks for your support, but we'd love to have more free time! :clock1:  If you really wish to support me, then that would be supporting [Polychromatic](https://polychromatic.app), one of the front-ends complementing this project. Refer to the [bottom of the README](https://github.com/polychromatic/polychromatic#donations) if you'd really want to drop a few pennies. :smile:

auipga, 2019-04-23:
Off topic:
I bought another device which is supported right now.
Is it true, that all apps only support lights but no key mapping, profiles and supershift?

lah7, 2019-04-23:
Yes, that's correct at this time.

If it's a keyboard, the 3 GUIs (Polychromatic, RazerGenie and razerCommander) support setting static colours per key, but each app may refer to that feature by different names, like "custom effect", "key mapping" or "application profile".

auipga, 2019-04-23:
Thanks for clarification.
I'm not the blink-blink kinda guy. I just need to change the key layout in a comfortable way.
Is there a solution you can recommend?

lah7, 2019-04-23:
I've heard from users that they've used [Keyboarding Master](https://sites.google.com/site/keyboardingmaster/) for remapping keys, but I haven't personally used it. [Pystromo](https://github.com/openrazer/openrazer/issues/167#issuecomment-452164735) would look promising, but is outdated and doesn't seem to work.

madmalkav, 2019-06-24:
I can confirm I have used Keyboarding Master in the past. Works mostly OK, being my only "serious" issue I cannot map "º" to any key of my Nostromo. That's the key left of "1" and up the Tab and it is quite frequent key used to play with ISO keyboards I think.

chabad360, 2020-04-19:
@auipga currently the only solution I've found that (seems to) work well is [tildearrow/input-modifer](https://github.com/tildearrow/input-modifier), the interface is a bit clunky, and it doesn't seem to be maintained any more. On the plus side, there's no scripting required, and it doesn't rely on Xinput (unlike Keyboarding Master) so it even work without XOrg.