openrazer/openrazer #817: Razer Goliathus Extended not mounting (Error Code 1)

Jcat4609, 2019-04-12:
Hey there, sorry for bad etiquette/formatiting/etc, not a huge github reporting person ^^;

I'm running Fedora 29 with a Blackwidow Chroma (with no issues), a Razer Basilisk, and a Razer Goliathus Extended (which is the problem child of this issue)

Recently (a couple weeks ago) I saw there was an update to OpenRazer which added support for my mousepad. I didn't end up finding it in my RazerGenie or my Polychromatic, so I took a little peek with `sudo journalctl -b | grep -i razer_mount`. Here's the relevant part of what I found:

```
Apr 12 13:46:29 localhost.localdomain razer_mount[826]: Driver razermousemat
Apr 12 13:46:29 localhost.localdomain razer_mount[830]: Driver razermousemat
Apr 12 13:46:29 localhost.localdomain razer_mount[827]: Driver razermousemat
Apr 12 13:46:29 localhost.localdomain razer_mount[832]: Device_ID 0003:1532:0C02.000B
Apr 12 13:46:29 localhost.localdomain razer_mount[829]: Device_ID 0003:1532:0C02.000A
Apr 12 13:46:29 localhost.localdomain razer_mount[834]: Modprobing razermousemat
Apr 12 13:46:29 localhost.localdomain razer_mount[833]: Device_ID 0003:1532:0C02.0009
Apr 12 13:46:29 localhost.localdomain razer_mount[835]: Modprobing razermousemat
Apr 12 13:46:29 localhost.localdomain systemd-udevd[686]: Process 'razer_mount razermousemat 0003:1532:0C02.000B' failed with exit code 1.
Apr 12 13:46:29 localhost.localdomain razer_mount[839]: Modprobing razermousemat
Apr 12 13:46:29 localhost.localdomain systemd-udevd[690]: Process 'razer_mount razermousemat 0003:1532:0C02.000A' failed with exit code 1.
Apr 12 13:46:29 localhost.localdomain systemd-udevd[695]: Process 'razer_mount razermousemat 0003:1532:0C02.0009' failed with exit code 1.
```

I took a peek at previous issues with razer_mount exiting with Error Code 1, and running `sudo modprobe razerfirefly; sleep 1; lsmod | grep razer` returns

```
razerfirefly           40960  0
razerkbd               69632  0
```
I thought I could be sneaky and type "razergol" and hit tab to autocomplete any things but nothing showed up hehe

If you peeps got any ideas, I'll test them as soon as I can - sorry if I reply late, NZT makes it inconvenient to talk to anyone who exists in any other country ever ^^;

z3ntu, 2019-04-13:
It seems that the dkms modules weren't rebuilt properly. Make sure to reboot and run `sudo dkms status` and post the output here

Jcat4609, 2019-04-13:
```
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/openrazer-driver/2.4.0/source/dkms.conf does not exist.
```

hehe well that seems to answer itself then huh

decided to try and run sudo dkms install openrazer-driver/2.4.0 but that also returns a lack of a dkms.conf too

z3ntu, 2019-04-13:
The current version is 2.5.0.

Maybe `sudo dkms remove openrazer-driver/2.4.0 --all` helps

Jcat4609, 2019-04-13:
`
[pasta@TheRig ~]$ sudo dkms remove openrazer-driver/2.4.0 --all
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/openrazer-driver/2.4.0/source/dkms.conf does not exist.
`
fascinating

you did bring to light to my mind that 2.5.0 is the latest version though, so i tried `sudo dkms install openrazer-driver/2.5.0` and that seemed to actually do something, will reboot and edit this post on how it goes

closed 2019-04-13 by Jcat4609

adueppen, 2020-02-09:
I've been having an issue related to this on my Fedora 31 system, the module works normally until I update my kernel. After that, I have to `dkms install` and `modprobe` it again, and then reboot. Running `dkms status` gives me this:
```
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/openrazer-driver/2.6.0/source/dkms.conf does not exist.
```
This seems rather odd since I'm using 2.7.0. I understand that this may be more of a DKMS question than anything but how would I go about fixing this?

z3ntu, 2020-02-13:
@adueppen Try `sudo dkms remove openrazer-driver/2.6.0 --all` ? Does that change anything?

adueppen, 2020-02-15:
@z3ntu that just gives me the same result; 
```
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/openrazer-driver/2.6.0/source/dkms.conf does not exist.
```
Same with `uninstall`.

z3ntu, 2020-02-16:
@adueppen Assuming you don't have anything other than openrazer in `/var/lib/dkms` you can run the following as root `dnf uninstall openrazer-kernel-modules-dkms` , `rm -rf /var/lib/dkms` and `dnf install openrazer-kernel-modules-dkms`

Hopefully that will get it working again.

adueppen, 2020-02-21:
It seems to all be working again, thank you for the help @z3ntu!