openrazer/openrazer #820: Support for [device name here]

Jinara, 2019-04-16:
### Device Info

* Name: <Blackwidow TE Chroma V2>
* Product Number: <RZ03-02191400-R311>
* Model Number: <Rz03-0219>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 008: ID 1532:0209 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0209 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
```
Please post the output of
    ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 abr 15 19:24 .
drwxr-xr-x 4 root root 620 abr 15 19:24 ..
lrwxrwxrwx 1 root root  10 abr 15 19:15 usb-Bison_Integrated_Camera_200901010001-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 abr 15 19:15 usb-KYE_SYSTEMS_CORP._Wired_Mouse-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 abr 15 19:15 usb-KYE_SYSTEMS_CORP._Wired_Mouse-mouse -> ../mouse2
lrwxrwxrwx 1 root root   9 abr 15 19:24 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 abr 15 19:24 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 abr 15 19:24 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 abr 15 19:24 usb-Razer_Razer_BlackWidow_Tournament_Edition_Chroma-if02-mouse -> ../mouse1

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

vjsingh-lite, 2019-08-05:
How do I get pcap on linux (arch)?

vjsingh-lite, 2019-08-05:
Interestingly `lsusb | grep "1532:` gives:
`Bus 001 Device 002: ID 1532:0209 Razer USA, Ltd BlackWidow Tournament Edition Chroma`

This device is supported, so maybe it should just work.

`~/.local/share/openrazer/logs/razer.log` shows:
`2019-08-05 16:17:07 | razer                          | CRITICAL | User is not a member of the plugdev group
2019-08-05 16:18:23 | razer                          | INFO     | Initialising Daemon (v2.6.0). Pid: 2239
2019-08-05 16:18:23 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2019-08-05 16:18:23 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
`

Unsure why it says user is not in `plugdev` group when it actually is.

z3ntu, 2019-08-05:
Thanks for noticing. 0209 is supported.

After adding your user to the plugdev group you either have to log out and back in or just reboot the PC.

closed 2019-08-05 by z3ntu

Jinara, 2019-09-09:
Hi @vjsingh-lite,

I've been trying, but my device is just not supported.

Name: Blackwidow TE Chroma V2

I saw the Name: Blackwidow TE Chroma without the V2 and Name: Blackwidow Chroma V2 without Tournament Edition.

Can you help me, please?

Thanks in advance!

Jinara, 2019-11-15:
Hi @vjsingh-lite , 

any chance to check this please?

Thank you very much!