openrazer/openrazer #823: The daemon didn't detect a device that is connected

manu212, 2019-04-28:
Hi,
I am using a Death Adder 2013 mouse which is detected by the daemon und my user has also been added to the plugdev group. OS: ArchLinux:

Linux detected connected devices but the daemon didn't. This could be either due to a permission problem or a kernel module problem.

Is the device not supported by razergenie? Thanks for any help to get the mouse visible in razergenie

Daemon output:
2019-04-28 12:15:25 | razer                          | DEBUG    | Device event [bind]: /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.4/4-1.4.1/4-1.4.1:1.2/0003:1532:0037.0003

z3ntu, 2019-04-28:
What's the output of `sudo dkms status` ?

BaronKrause, 2019-05-17:
I got I think the same message and solved it by following another person's advice about simply unplugging it and replugging it in (then restart the daemon or just tell it to retry from the app indicator context menu).

Don't know if its a Deathadder 2013 quirk, but often even after that It will come up with a red indicator and I will need to unplug and replug it in, then retry on the daemon icon.

closed 2019-05-25 by z3ntu